arduino/ArduinoCore-samd #2: .... Reset not supported for this CPU.

bobricius, 2015-06-16:
after upload to board via native port I have message 

Verify successful
done in 0.047 seconds
CPU reset.
Reset not supported for this CPU.

I need manually reset board

CPU is Atsamd21E18 is specially unsupported E version?
Is problem with bootloader or upload software?

cmaglie, 2015-06-16:
It may be that the CPU ID for Atsamd21E18 is not listed inside bossac. May you retry to upload with the verbose upload enabled from the preferences and copy here the output?

cmaglie, 2015-06-16:
Maybe this patch on bossac can help you?

https://github.com/shumatech/BOSSA/pull/12/files

bobricius, 2015-06-17:
Thanks maybe it can help but I do not know how to use this patch.

cmaglie, 2015-06-19:
@bobricius 
I've just pushed a test build here: https://github.com/shumatech/BOSSA/pull/12#issuecomment-113498651

let me know if it's working for you

bobricius, 2015-06-19:
YES.WORKING !!!! YOU ARE GREAT !!!

cmaglie, 2015-06-19:
Thanks, the updated bossac will be available on the next release of the core.

closed 2015-06-19 by cmaglie