arduino/ArduinoCore-samd #6: RX/TX led not blinking

cmaglie, 2015-06-18:
Update bootloader and USB-CDC driver in order to turn on RX/TX leds when there is serial activity.

http://forum.arduino.cc/index.php?topic=330585.msg0

cmaglie, 2015-06-18:
Related to #1

ladyada, 2016-02-14:
here's some CDC USB LED implement code you can use to try this out:
https://github.com/adafruit/ArduinoCore-samd/commit/440afc570d64ae96f1210723c0479e1704ee3368
tested on a zero (but of course, plz also test :)

sandeepmistry, 2016-03-04:
@ladyada thanks for the link :)

I've opened PR  #121 with something similar, it will make the LED's blink for all native USB traffic (including CDC), just like the Leonardo.

rory-littleBits, 2016-03-07:
#121 works. It blinks the USB TX and RX LEDs during serial communication. I tested the directions above. However, this doesn't address the issue where the LEDs don't blink when uploading a new sketch. I imagine this will need to be added to the Arduino Zero bootloader.

sandeepmistry, 2016-03-07:
@rory-littleBits great thanks for trying it out!

> I imagine this will need to be added to the Arduino Zero boot loader.

Yes, #121 only handles the non-bootloader side, #1 is open to track this feature in the boot loader.

cmaglie, 2016-04-19:
Solved by #121

closed 2016-04-19 by cmaglie