arduino/ArduinoCore-samd #9: Ethernet shield examples won't currently compile

gh-megabit, 2015-06-21:
The ethernet library for Wiznet W5100 appears to be written to support either the AVR series or the Due.

As it is, the code treats the Zero like a Due, but the SPI library is missing the function:

`void beginTransaction(uint8_t pin, SPISettings settings);`

I'm not sure how best to approach this for Zero. It may require changes to SPI, Ethernet or both.

See: http://forum.arduino.cc/index.php?topic=331074.0

cmaglie, 2015-06-22:
Pushed a fix for that in https://github.com/arduino-libraries/Ethernet, may you give it a try before releasing via Library Manager?

gh-megabit, 2015-06-22:
Thanks! I can confirm that this compiles for me now.

Unfortunately I won't be able to test functionality for a day or two.

cmaglie, 2015-07-18:
This has been tested and reported as working. Thanks!

closed 2015-07-18 by cmaglie