arduino/ArduinoCore-samd #11: Arduino Zero <avr/sleep.h>

ffissore, 2015-06-22:
_From @mitko9000 on June 22, 2015 7:31_

When trying to compile nordic nrf8001 arm support library it says missing <avr/sleep.h>
is that some kind of a issue or  just this miscrocontroller is not supporting this library ?

_Copied from original issue: arduino/Arduino#3393_

mitko9000, 2015-06-22:
it says missing avr/sleep.h

cmaglie, 2015-06-23:
You can easily overcome the error by moving the avr-specific `avr/sleep.h` include like this:

```diff
diff --git a/hal_aci_tl.cpp b/hal_aci_tl.cpp
index e10e768..f2b3249 100644
--- a/hal_aci_tl.cpp
+++ b/hal_aci_tl.cpp
@@ -26,18 +26,17 @@
 #include <SPI.h>
 #include "hal_platform.h"
 #include "hal_aci_tl.h"
 #include "aci_queue.h"
-#include <avr/sleep.h>
 
 /*
 PIC32 supports only MSbit transfer on SPI and the nRF8001 uses LSBit
 Use the REVERSE_BITS macro to convert from MSBit to LSBit
 The outgoing command and the incoming event needs to be converted
 */
 //Board dependent defines
 #if defined (__AVR__)
-    //For Arduino add nothing
+    #include <avr/sleep.h>^M
 #elif defined(__PIC32MX__)
     //For ChipKit as the transmission has to be reversed, the next definitions have to be added
     #define REVERSE_BITS(byte) (((reverse_lookup[(byte & 0x0F)]) << 4) + reverse_lookup[((byte & 0xF0) >> 4)])
     static const uint8_t reverse_lookup[] = { 0, 8,  4, 12, 2, 10, 6, 14,1, 9, 5, 13,3, 11, 7, 15 };
```

On the other end we can provide an "empty" `avr/sleep.h` file in the Zero's core to provide basic compatibility, even if this will reduce pressure to upstream authors to fix the library.

BTW even with this fix in place I get a bunch of "duplicate definition" errors:

```
tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o: In function `hal_aci_tl_debug_print(bool)':
/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:254: multiple definition of `hal_aci_tl_debug_print(bool)'
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o:/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:254: first defined here
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o: In function `hal_aci_tl_pin_reset()':
/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:258: multiple definition of `hal_aci_tl_pin_reset()'
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o:/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:258: first defined here
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o: In function `hal_aci_tl_event_peek(hal_aci_data_t*)':
/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:282: multiple definition of `hal_aci_tl_event_peek(hal_aci_data_t*)'
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o:/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:282: first defined here
/tmp/build1848335976567567191.tmp/nrf8001_arm_support/hal_aci_tl.cpp.o: In function `hal_aci_tl_event_get(hal_aci_data_t*)':
/home/megabug/Code/Arduino/libraries/nrf8001_arm_support/hal_aci_tl.cpp:297: multiple definition of `hal_aci_tl_event_get(hal_aci_data_t*)'
```

If I try to compile for a Uno instead I get:

```
ble_broadcast.ino:64:23: error: setup_msgs causes a section type conflict with __c
In file included from /home/megabug/Code/arduino/build/linux/work/hardware/arduino/avr/cores/arduino/Arduino.h:28:0,
                 from /home/megabug/Code/arduino/build/linux/work/hardware/arduino/avr/libraries/SPI/SPI.h:17,
                 from ble_broadcast.ino:42:
/home/megabug/Code/arduino/build/linux/work/hardware/arduino/avr/cores/arduino/WString.h:38:74: note: '__c' was declared here
 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
                                                                          ^
ble_broadcast.ino:226:24: note: in expansion of macro 'F'
ble_broadcast.ino:79:23: warning: 'aci_cmd' defined but not used [-Wunused-variable]
setup_msgs causes a section type conflict with __c
```

The above I guess that is due to missing `const` and deprecated `prog_uchar`.

mitko9000, 2015-06-24:
To compile for ARM due you have to use this source as BLE library -> https://github.com/Cheong2K/ble-sdk-arduino/tree/RBL/src

sandeepmistry, 2016-01-19:
Closing, as this can be addressed in the upstream library in a similar manner to as how SAM supported was added in https://github.com/NordicSemiconductor/ble-sdk-arduino/commit/1bc1749c6e22eaa2cc9938e256cfa68d5998fd82.

closed 2016-01-19 by sandeepmistry