arduino/ArduinoCore-samd #28: the pin mode is changed to INPUT mode

Makeblock-official, 2015-08-23:
on file wiring.c line 87(Fun init()) Setup all pins (digital and analog) in INPUT mode (default is nothing) 
Pin mode should not be set.This line should comment.

on file wiring_digital.c line 81 (Fun digitalWrite()) Enable pull-up resistor.
This line should comment.

The pin mode will be initialized when construct the object.

For example:
class LED
{
  public:
    LED(int pin);
    void low();
    void high();
  private:
    int led_pin;
};

LED::LED(int pin)
{
  led_pin = pin;
  pinMode(led_pin, OUTPUT) ;
}

void LED::low()
{
  digitalWrite(led_pin, LOW);
}
void LED::high()
{
  digitalWrite(led_pin, HIGH);
}

LED led(13);
void setup() {
}

void loop() {
  led.high();
  delay(1000);
  led.low();
  delay(1000);
}

Since the pin mode is changed to INPUT,led.high() is set to input pull-up and led.low() is set to input pull-down.So the led will be very dark.

aethaniel, 2015-09-04:
Indeed, the pull-up should be set only on a digitalWrite(pin, HIGH) when pin is in INPUT mode.

Makeblock-official, 2015-10-10:
It is not correct, I mean, Pin mode should not be set at initialization.

the Fun void init( void )
  // Setup all pins (digital and analog) in INPUT mode (default is nothing)
  for ( ul = 0 ; ul < NUM_DIGITAL_PINS ; ul++ )
  {
	  pinMode( ul, INPUT ) ;//This line should comment.
  }

void digitalWrite( uint32_t ulPin, uint32_t ulVal )
{
  // Handle the case the pin isn't usable as PIO
  if ( g_APinDescription[ulPin].ulPinType == PIO_NOT_A_PIN )
  {
    return ;
  }

the Fun void digitalWrite( uint32_t ulPin, uint32_t ulVal )
  // Enable pull-up resistor
  PORT->Group[g_APinDescription[ulPin].ulPort].PINCFG[g_APinDescription[ulPin].ulPin].reg=(uint8_t)(PORT_PINCFG_PULLEN) ;//This line should comment.

aethaniel, 2015-10-10:
@Makeblock-official, can you put comments in the pull request itself on the lines which disturb you, please?

WestfW, 2015-12-28:
Having all pins start out as inputs is compatible with the AVR Arduinos...

sandeepmistry, 2016-01-20:
Closing as setting all pins to INPUT is desired, to maintain compatibility AVR based boards as mentioned by @WestfW above.

#94 is open the track @Makeblock-official's concern from https://github.com/arduino/ArduinoCore-samd/issues/28#issuecomment-147111916.

> the Fun void digitalWrite( uint32_t ulPin, uint32_t ulVal )
> // Enable pull-up resistor
> PORT->Group[g_APinDescription[ulPin].ulPort].PINCFG[g_APinDescription[ulPin].ulPin].reg=(uint8_t)(PORT_PINCFG_PULLEN) ;//This line should comment.

closed 2016-01-20 by sandeepmistry