arduino/ArduinoCore-samd #36: No Valid JTAG Error

Proclivis, 2015-09-07:
I got a new Zero from arduino.cc, followed the getting started (Windows), and succeeded running a simple program, using the Programming Port. But on 32 or 64 bit Ubuntu, I can't upload the bits. Both cases give the error below. I also used the hourly version on 64 bit and got the same result.

These same machines have been used for an Uno for quite some time, so there should not be any generic problems with the USB, /dev/tty# etc. The port also shows up properly, and with an addition to the rules file, I can pop the console without error.

Just in case it mattered, I installed bossa and bossa-cli as follows:
sudo apt-get install bossa-cli
sudo apt-get install bossa

I did not expect this to matter based on the error, just pugging possible holes.

Perhaps there is some magic needed in a configuration file somewhere?

Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-18:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.

agdl, 2015-09-07:
please try to execute the IDE as root. If it works you have to add a rule to udev!

Proclivis, 2015-09-07:
Hm, I had already added a rule, but sudo fixed it. After some trial and error, this worked:

SUBSYSTEMS=="usb", ATTRS{idProduct}=="2157", ATTRS{idVendor}=="03eb", MODE="0666", SYMLINK+="ttyACM%n”

Seems that the MODE is required, even though the file attributes are 777. Strange, but it works now.

My other entries are:

UBSYSTEMS=="usb", ATTRS{idProduct}=="0043", ATTRS{idVendor}=="2341", SYMLINK+="ttyACM%n"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="6001", ATTRS{idVendor}=="0403", SYMLINK+="ttyACM%n”

And they worked without MODE.

Thanks,

Mike

Proclivis, 2015-09-08:
Perms was problem.

closed 2015-09-08 by Proclivis