arduino/ArduinoCore-samd #37: SMBus Support

Proclivis, 2015-09-08:
The SERCOM file does not support repeated start, therefore SMBus transactions are not possible in Wire.

Can someone explain how to compile the serial library and put it into play? I would be happy to try to add support for repeated start. Im working on Ubuntu, so it would be nice to compile with a gcc toolchain, etc.

I work on SMBus stuff a lot, so I don't need handholding on the protocol. When done, the LTC PMBus library will eventually release to the public for anyone to use. Today it only works on Uno, Mega, etc.

legomanww, 2015-11-10:
I've made the modifications I believe are necessary to enable repeated starts.

Here's a link to my fork commit: [Commit](https://github.com/legomanww/ArduinoCore-samd/commit/6747af02eb0e32a3845f9f85ff5043fb8a7b2ea8)

And a scope shot of it in action:
[image]

I can submit a pull request if someone else wants to test and verify the changes.

sandeepmistry, 2015-11-17:
@legomanww thanks for sharing your change!

I've pulled it into my branch and modified it to wait for a bus state of idle or owner. This removes the need to track the repeated start state. See the following commit: https://github.com/sandeepmistry/ArduinoCore-samd/commit/9ae66330dc894afaa74d37e0a35fad4b534f17a3

Would you be able to test my branch out and see what the scope shows before I submit a pull request?
https://github.com/sandeepmistry/ArduinoCore-samd/tree/wire-master-repeated-starts

legomanww, 2015-11-18:
Looks OK, except it doesn't compile as-is.  Line 71 on Wire.cpp still has a reference to repeatedStart.

sandeepmistry, 2015-11-18:
@legomanww thanks for trying it out!

I've corrected and created pull request #66.

PR #66 (2016-01-13): Correct Wire master repeated start behaviour
The ``stopBit`` argument in ``TwoWire::requestFrom`` and ``TwoWire::endTransmission`` was being ignored.

Based on @legomanww's patch https://github.com/legomanww/ArduinoCore-samd/commit/6747af02eb0e32a3845f9f85ff5043fb8a7b2ea8.

This resolves #37.
closed 2016-01-13 by cmaglie