arduino/ArduinoCore-samd #48: HardwareSerial BUG Rx pin floating input

Makeblock-official, 2015-10-18:
Rx pin should be set to the pull-input at begin

fix this bug as follows
file Uart.cpp line 37
void Uart::begin(unsigned long baudrate, uint8_t config)
{
  pinPeripheral(uc_pinRX, g_APinDescription[uc_pinRX].ulPinType);
  pinPeripheral(uc_pinTX, g_APinDescription[uc_pinTX].ulPinType);

  sercom->initUART(UART_INT_CLOCK, SAMPLE_RATE_x16, baudrate);
  sercom->initFrame(extractCharSize(config), LSB_FIRST, extractParity(config), extractNbStopBit(config));
  sercom->initPads(uc_padTX, uc_padRX);
  pinMode(uc_pinRX,INPUT_PULLUP);//Rx pin set INPUT_PULLUP mode,
  sercom->enableUART();
}

file SERCOM.cpp line 118
void SERCOM::clearStatusUART()
{
  //Reset (with 0) the STATUS register
  //sercom->USART.STATUS.reg = SERCOM_USART_STATUS_RESETVALUE;
   sercom->USART.STATUS.reg = 0xFF;//Writing a zero to this bit has no effect.,Writing a one to this bit will clear it.
}

Makeblock-official, 2015-10-18:
Uart::Uart(SERCOM *_s, uint8_t _pinRX, uint8_t _pinTX, SercomRXPad _padRX, SercomUartTXPad _padTX)
{
  sercom = _s;
  uc_pinRX = _pinRX;
  uc_pinTX = _pinTX;
  uc_padRX=_padRX ;
  uc_padTX=_padTX;
pinMode(uc_pinRX,INPUT_PULLUP);//Rx pin set INPUT_PULLUP mode
}
This method is not effective,the program runs to main(),will change the pin mode to INPUT

file wiring.c line 87
void init( void )
{
  uint32_t ul ;

  // Set Systick to 1ms interval, common to all Cortex-M variants
  if ( SysTick_Config( SystemCoreClock / 1000 ) )
  {
    // Capture error
    while ( 1 ) ;
  }

  // Clock PORT for Digital I/O
//	PM->APBBMASK.reg |= PM_APBBMASK_PORT ;
//
//  // Clock EIC for I/O interrupts
//	PM->APBAMASK.reg |= PM_APBAMASK_EIC ;

  // Clock SERCOM for Serial
  PM->APBCMASK.reg |= PM_APBCMASK_SERCOM0 | PM_APBCMASK_SERCOM1 | PM_APBCMASK_SERCOM2 | PM_APBCMASK_SERCOM3 | PM_APBCMASK_SERCOM4 | PM_APBCMASK_SERCOM5 ;

  // Clock TC/TCC for Pulse and Analog
  PM->APBCMASK.reg |= PM_APBCMASK_TCC0 | PM_APBCMASK_TCC1 | PM_APBCMASK_TCC2 | PM_APBCMASK_TC3 | PM_APBCMASK_TC4 | PM_APBCMASK_TC5 ;

  // Clock ADC/DAC for Analog
  PM->APBCMASK.reg |= PM_APBCMASK_ADC | PM_APBCMASK_DAC ;

  // Setup all pins (digital and analog) in INPUT mode (default is nothing)
  for ( ul = 0 ; ul < NUM_DIGITAL_PINS ; ul++ )
  {
	  pinMode( ul, INPUT ) ;//change the pin mode to INPUT
  }

cmaglie, 2015-10-20:
@Makeblock-official sorry, I don't understand.
May you submit your patch as a pull-request, or at least explain better what's happening?

sandeepmistry, 2016-01-25:
The RX pin does not have pull-up mode enabled by default to maintain compatibility with the AVR core.

Once something like #101 is merged, you'll be able to enable pull-ups for RX0 in your sketch if needed.

I'm closing this for now. @Makeblock-official why do you need to enable pull up on RX0?

closed 2016-01-25 by sandeepmistry

mamama1, 2022-06-15:
RX needs to be INPUT_PULLUP in my opinion because in disconnected state it will pick up noise and also floating pins are very bad for power consumption during sleep (not just bad, but horrible! floating around 100µA instead of 3µA).

can someone please pursue this issue.

mamama1, 2022-06-18:
has there been any progress? floating pins are not too good if you are trying to build a low power thingo.