arduino/ArduinoCore-samd #49: Press the reset button does not start  bootloader

Makeblock-official, 2015-10-18:
If the user's program was wrong, the board can not start bootloader. USB can not update the firmware. Only update the firmware via the programming port.

For example:

void setup() {
  pinMode(100,OUTPUT);
}

void loop() {
}

Because this pin number 100 is not available, the program will go wrong, bootloader can not start

Makeblock-official, 2015-10-18:
bootloader firmware can refer arduino leonardo board, press the reset button to start bootloader

cmaglie, 2015-10-20:
Pressing the reset button twice should start the bootloader, may you give it a try?

sandeepmistry, 2016-01-20:
Closing due to lack of activity, and @cmaglie has answered how to start the bootloader in https://github.com/arduino/ArduinoCore-samd/issues/49#issuecomment-149726299.

closed 2016-01-20 by sandeepmistry