arduino/ArduinoCore-samd #52: tone function random crashes

PaulStoffregen, 2015-10-21:
This sketch will crash after several minutes running on Arduino Zero.  Serial output stops, and the frequency on pin 5 becomes approx 258 kHz.

    void setup() {
      Serial.begin(115200);
      while (!Serial && millis() < 1500) ;
      delay(10);
      Serial.println("Tone Issue Test");
      delay(20);
    }

    void loop() {
      int frequency = 1500 + random(800);
      tone(5, frequency);
      Serial.println(frequency);
      delayMicroseconds(random(10000));
    }

aethaniel, 2015-10-22:
Hi @PaulStoffregen, not yet as I didn't implement this API. I will check the code with @agdl but not before next Wednesday, unfortunately.

PaulStoffregen, 2015-10-22:
I'm in no hurry.  Just wanted to make sure you knew about the crashing.

agdl, 2015-10-22:
@PaulStoffregen try this please https://github.com/arduino/ArduinoCore-samd/tree/toneFix

Your sketch has been running for 2 hours

closed 2016-02-15 by cmaglie