arduino/ArduinoCore-samd #56: writing LOW to a digital input pin blocks subsequent digitalRead attempts

soundanalogous, 2015-10-24:
I noticed while trying to add Arduino Zero support for Firmata that the following line fails: https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L266.
After that point, digital input will not work on that pin (it's not permanently broken as in fried input pin or anything like that though). This has never been an issue for other architectures.

One one hand this points out that I need to update the Firmata implementation to use `INPUT_PULLUP`, something I should have probably added long ago. On the other hand this is an inconsistency so just calling that out.

This may be the same issue as #28.

soundanalogous, 2015-10-24:
The following does not work on SAMD boards either:
```c++
byte pin = 2;
pinMode(pin, INPUT);
digitalWrite(pin, HIGH); // enable pull-ups

// some time later
digitalRead(pin); // does not work
```

However this works (as expected):
```c++
byte pin = 2;
pinMode(pin, INPUT_PULLUP);

// some time later
digitalRead(pin); // value is readable
```

Personally I don't mind this and think it's the best way to go forward, but it was unexpected as it does work on all non SAMD architectures (including SAM) so the Arduino pinMode and digitalRead documentation should be updated for Zero if this change is intentional.

aethaniel, 2015-10-30:
this patch could maybe fix your issue: https://github.com/arduino/ArduinoCore-samd/pull/41

sandeepmistry, 2016-01-21:
@soundanalogous AVR compatible behaviour should be maintained if possible.

I've submitted #101 to resolve the issue you mentioned in https://github.com/arduino/ArduinoCore-samd/issues/56#issuecomment-150857149. Please try it out if you have time.

PR #101 (2016-07-08): Enable digitalRead on output pins, only enable pull-up in digitalWrite if pin is not output
Resolves #56 and #94.

* ```pinMode```:
 * enable input mode of output pins, to support using ```digitalRead``` to read current value (maintains AVR behaviour).
 * disable pull-up on output mode
* ```digitalWrite```:
 * only enable pull-up if pin is not in output mode and ```HIGH``` is passed in, disable pull-up if ```LOW``` passed in. (This is similar to #41, but it did not set OUT which is needed to enable pull-up - as per table 22-1 in the SAMD21 datasheet).
 * simplify some duplicated code, by using [... cut]
closed 2016-07-08 by agdl