arduino/ArduinoCore-samd #62: 1200bps reset unreliable

pixelpixi, 2015-11-11:
When uploading using the native usb port, uploading fails every other time. I think that after the first failed upload the board is left running the bootloader, which is what allows the second attempt to succeed. This happens both with the arduino zero (native usb port) and my own modulo controller. It does not happen will non-samd boards such as the arduino micro.

Here's the sequence of events that I'm seeing:

1) Press upload (using the Blink sketch for this test)
2) 1200bps reset occurs
3) upload port is found
4) bossac FALS with the error: "No device found on cu.usbmodemXXXX"
5) Press upload
6) 1200bps reset occurs
7) upload port is found
8) bossac succeeds.

Here's the output from a failed upload:

    Sketch uses 10,988 bytes (4%) of program storage space. Maximum is 262,144 bytes.
    Forcing reset using 1200bps open/close on port /dev/cu.usbmodem1411
    PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem1411, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem1411, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
    PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
    PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
    PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem1411, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem1411, } => {/dev/cu.usbmodem1411, /dev/tty.usbmodem1411, }
    Found upload port: /dev/cu.usbmodem1411
    /Users/ekt/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino/bossac -i -d --port=cu.usbmodem1411 -U true -i -e -w -v /var/folders/7z/hphfkcsx0nq7cdntfsf967g00000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/Blink.ino.bin -R 
    No device found on cu.usbmodem1411

Is it possible that the bootloader takes too long to initialize the port and bossac runs before the port is ready?

Also I think this may have started happening after upgrading to OS X El Capitan. I haven't tested it on other operating systems but can do so if that's helpful.

Thanks a lot!

sandeepmistry, 2016-01-21:
I am also seeing this behaviour with the OS X 10.11.3, I think it is something do to with OS X USB CDC driver changes.

When reading/writing to the programming port on the Zero I've seen the first byte getting drop after the 2nd sketch upload, but first upload after plugging in the board works all the time. I was not seeing this behaviour with OS X 10.10.x. It may or may not be related to this.

sandeepmistry, 2016-01-25:
This might be timing related, if I lower the tick count passed into ```initiateReset``` in ```CDC_Setup``` (cores/arduino/USB/CDC.cpp) from 250 to 125 I no longer see the error (loading the Blink example sketch).

I also tried disabling the CDC endpoint and pressing the reset button twice to enter boot loader mode, and did not see this issue

> Is it possible that the bootloader takes too long to initialize the port and bossac runs before the port is ready?

@pixelpixi this seems likely. @aethaniel any thoughts on this?

This might be unrelated, but the OS X logs always have to following message anytime the boot loader is launched:

```
2016-01-25 4:33:08.000 PM kernel[0]: 129020.020709 IOUSBHostDevice@14200000: IOUSBHostDevice::getDescriptorGated: type 0x03 index 0 length 2 completed with 0xe0005000 and bytesTransferred 0
2016-01-25 4:33:08.000 PM kernel[0]: 129020.022294 IOUSBHostDevice@14200000: IOUSBHostDevice::getDescriptorGated: type 0x03 index 0 length 2 completed with 0xe0005000 and bytesTransferred 0
```

and the following after a sketch starts running:
```
2016-01-25 4:34:24.940 PM VDCAssistant[214]: VDCAssistant:  Found a camera (0x142000002341804d) , but was not able to start it up (0x0 -- (os/kern) successful)
2016-01-25 4:34:24.940 PM VDCAssistant[214]: VDCAssistant:  Found a camera (0x142000002341804d) , but was not able to start it up (0x0 -- (os/kern) successful)
```

sandeepmistry, 2016-01-26:
I am no longer see this error when trying @facchinm's improvements in https://github.com/arduino/Arduino/pull/4482.

@pixelpixi it would be great if you could also downloaded the IDE build from https://github.com/arduino/Arduino/pull/4482#issuecomment-174617069 and try it out on your Mac.

pixelpixi, 2016-01-26:
I tried it and it works! Hurray! Thanks a lot.

pixelpixi, 2016-01-26:
I may have spoken too soon. It appears to work consistently when connected directly to my computer, but when USB hubs are between my computer and the board it still fails. Perhaps the hubs affect how long it takes for the port to be initialized?

NicoHood, 2016-01-30:
Looks like this:
https://github.com/abcminiuser/lufa/issues/44

sandeepmistry, 2016-02-01:
@pixelpixi here's another IDE patch for you to try out: https://github.com/arduino/Arduino/pull/4515#issuecomment-178206278

Without the change the IDE was only waiting 0.5s for the upload port to disappear and re-appear. I found it was taking ~850ms with OS X 10.11.3 and the Arduino Zero. The IDE patch bumps the wait period to 5s on all platforms.

pixelpixi, 2016-02-01:
Okay, this is weird. With the new build  ( PR-4515-BUILD-497 ) it works more reliably (though not 100% of the time) with USB hubs in between, but now fails consistently again when connected directly to the computer.

sandeepmistry, 2016-02-02:
@pixelpixi thanks for trying out the change.

> it works more reliably (though not 100% of the time) with USB hubs in between

I tested with a custom build of ```bossac```, occasionally I get ```Resource busy``` error on open. However, it is much more reliable, without the IDE change ```bossac``` reported ```No such file or directory``` on open.

>  but now fails consistently again when connected directly to the computer.

Just wanted to confirm this happens with a Zero board and the boot loader from master?

I can't reproduce with the boot loader from master, only with the modified boot loader from #106.

pixelpixi, 2016-02-03:
I was testing with a modulo controller (which uses the arduino zero bootloader from a while back).

I can test on a zero board using the bootloader from master if you think that's the problem, though. If I install the hourly build of the Arduino SAMD Boards package and then do "Burn Bootloader" will I get the right one?

By the way, is there an easy way to use a local repo of the samd core? I always install the latest version, then move it out of the way, then create a symlink. (ie, mv 1.6.2 1.6.2.old; ln -s ~/ArduinoSamdCore 1.6.2). Maybe there's a better way to do it?

Thanks!

pixelpixi, 2016-02-03:
I tried installing the hourly build and burning the bootloader to the zero. It's still failing unfortunately.

I'm unsure if I have the right version though. Is the version number that's printed when uploading the bootloader version? It says, "Version      : v2.0 [Arduino:XYZ] Oct 10 2015 15:57:45" which sounds pretty old.

sandeepmistry, 2016-02-12:
> By the way, is there an easy way to use a local repo of the samd core? I always install the latest version, then move it out of the way, then create a symlink. (ie, mv 1.6.2 1.6.2.old; ln -s ~/ArduinoSamdCore 1.6.2). Maybe there's a better way to do it?

I have a symlink to my Git checkout in ```~/Documents/Arduino/hardware/arduino/samd``` (you might have to create the hardware and arduino folders).

> I tried installing the hourly build and burning the bootloader to the zero. It's still failing unfortunately.

Ok, I'll spend some time looking at the boot loader, and let you know if I have a new version you can test.

Does your Mac have USB 2.0 or USB 3.0 ports?

> I'm unsure if I have the right version though. Is the version number that's printed when uploading the bootloader version? It says, "Version : v2.0 [Arduino:XYZ] Oct 10 2015 15:57:45" which sounds pretty old.

```Version      : v2.0 [Arduino:XYZ] Oct 10 2015 15:57:45``` matches the boot loader I was testing with.

sandeepmistry, 2016-02-29:
@pixelpixi here's another IDE build for you to try: https://github.com/arduino/Arduino/pull/4515#issuecomment-190322450

I found if a ``/dev/cu.usbmodem*`` was opened too quickly after the IDE detected it, a "Resource Busy" error is returned on open. The latest commit, adds an OS X specific delay after discovering the port.

sandeepmistry, 2016-04-04:
I'm closing this https://github.com/arduino/Arduino/pull/4515 has been merged and is available in the latest hourly build.

closed 2016-04-04 by sandeepmistry