arduino/ArduinoCore-samd #79: digitalRead() broken on pins in output mode.

WestfW, 2015-12-28:
Apparently "pinMode(pin, OUTPUT)" will clear the INEN bit for the GPIO pin in question, preventing digitalRead() (or direct port access) from being able to read the current output state of an output pin, and breaking trivial statements like:
    digitalWrite(13, ! digitalRead(13));  // toggle LED
As far as I can tell, there should never be a reason to disable input on the GPIO pins (PORT_PINCFG_INEN bit cleared in pincfg), except *maybe* when the pin is used for analog input.

http://forum.arduino.cc/index.php?topic=367517

WestfW, 2015-12-28:
(also be careful not to clear INEN when setting PULLEN)

sandeepmistry, 2016-01-21:
@WestfW agreed, I've submitted #101 to resolve this, please review or test it out if you have time.

sandeepmistry, 2016-07-08:
Closing since #101 is merged now.

closed 2016-07-08 by sandeepmistry