arduino/ArduinoCore-samd #80: USB host library conflicts with Due USB host library

bbx10, 2015-12-29:
[image]

The library manager recommends updating the USB host library to 1.0.4 but this version does not work on Zero. If the update is done, the keyboard example does not compile. I do not see any way to suppress this message.

IDE 1.6.7 and SAMD board version hourly build.

USB host library 1.0.4 works fine on Due so there is nothing wrong with the library.

aethaniel, 2016-01-13:
hi @bbx10, can you activate the compilation trace in files->preferences and put here the lines presenting the issue from trace window, please?

bbx10, 2016-01-13:
@aethaniel OK here is the compile log for the USB keyboard example build failure after upgrading USBHost to 1.0.4. The example compiled and ran before the upgrade using the built-in USBHost library 1.0.0.
```
/home/zero/arduino-1.6.7/arduino-builder -dump-prefs -logger=machine -hardware "/home/zero/arduino-1.6.7/hardware" -hardware "/home/zero/.arduino15/packages" -tools "/home/zero/arduino-1.6.7/tools-builder" -tools "/home/zero/arduino-1.6.7/hardware/tools/avr" -tools "/home/zero/.arduino15/packages" -built-in-libraries "/home/zero/arduino-1.6.7/libraries" -libraries "/home/zero/Arduino/libraries" -fqbn=arduino:samd:arduino_zero_edbg -ide-version=10607 -build-path "/tmp/build319ab28555b0a5a8314cb58672fe1a2b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/zero/Arduino/libraries/USBHost/examples/KeyboardController/KeyboardController.ino"
/home/zero/arduino-1.6.7/arduino-builder -compile -logger=machine -hardware "/home/zero/arduino-1.6.7/hardware" -hardware "/home/zero/.arduino15/packages" -tools "/home/zero/arduino-1.6.7/tools-builder" -tools "/home/zero/arduino-1.6.7/hardware/tools/avr" -tools "/home/zero/.arduino15/packages" -built-in-libraries "/home/zero/arduino-1.6.7/libraries" -libraries "/home/zero/Arduino/libraries" -fqbn=arduino:samd:arduino_zero_edbg -ide-version=10607 -build-path "/tmp/build319ab28555b0a5a8314cb58672fe1a2b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/zero/Arduino/libraries/USBHost/examples/KeyboardController/KeyboardController.ino"
Warning: platform.txt from core 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
"/home/zero/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/cores/arduino" "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/variants/arduino_zero" "/tmp/build319ab28555b0a5a8314cb58672fe1a2b.tmp/sketch/KeyboardController.ino.cpp" -o "/dev/null"
"/home/zero/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/cores/arduino" "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/variants/arduino_zero" "-I/home/zero/Arduino/libraries/USBHost/src" "/tmp/build319ab28555b0a5a8314cb58672fe1a2b.tmp/sketch/KeyboardController.ino.cpp" -o "/dev/null"
"/home/zero/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/cores/arduino" "-I/home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/variants/arduino_zero" "-I/home/zero/Arduino/libraries/USBHost/src" "/home/zero/Arduino/libraries/USBHost/src/KeyboardController.cpp" -o "/dev/null"
"/home/zero/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/zero/.arduino15/packages/arduino/tools/CMSIS/4 [... cut ...]
/home/zero/Arduino/libraries/USBHost/src/Usb.h:193:35:   required from here
/home/zero/Arduino/libraries/USBHost/src/address.h:290:4: error: 'struct UsbDevice' has no member named 'address'
    if (thePool[i].address)
    ^
Multiple libraries were found for "KeyboardController.h"
 Used: /home/zero/Arduino/libraries/USBHost
 Not used: /home/zero/.arduino15/packages/arduino/hardware/samd/1.6.2/libraries/USBHost
Using library USBHost at version 1.0.4 in folder: /home/zero/Arduino/libraries/USBHost 
exit status 1
Error compiling.

```

sandeepmistry, 2016-04-04:
Now that https://github.com/arduino/arduino-builder/pull/123 has been merged, this is resolved in the hourly IDE builds.

closed 2016-04-04 by sandeepmistry

bbx10, 2016-04-04:
I tried hourly build 2016-04-02 12:45 with IDE 1.6.8. The keyboard example compiles fine using the built in USBHost 1.0.0.

Library manager says USBHost is updatable to 1.0.5 so I update it. The keyboard example fails to compile.

Must the USBHost library maintainers make a change in addition to the patch?

```
Arduino: 1.6.8 (Linux), Board: "Arduino/Genuino Zero (Programming Port)"

/home/me/arduino-1.6.8/arduino-builder -dump-prefs -logger=machine -hardware "/home/me/arduino-1.6.8/hardware" -hardware "/home/me/.arduino15/packages" -tools "/home/me/arduino-1.6.8/tools-builder" -tools "/home/me/arduino-1.6.8/hardware/tools/avr" -tools "/home/me/.arduino15/packages" -built-in-libraries "/home/me/arduino-1.6.8/libraries" -libraries "/home/me/Arduino/libraries" -fqbn=arduino-beta:samd:arduino_zero_edbg -ide-version=10608 -build-path "/tmp/buildb6ea5fcceab80e8a4f56da35730111a7.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "/home/me/Arduino/libraries/USBHost/examples/KeyboardController/KeyboardController.ino"
/home/me/arduino-1.6.8/arduino-builder -compile -logger=machine -hardware "/home/me/arduino-1.6.8/hardware" -hardware "/home/me/.arduino15/packages" -tools "/home/me/arduino-1.6.8/tools-builder" -tools "/home/me/arduino-1.6.8/hardware/tools/avr" -tools "/home/me/.arduino15/packages" -built-in-libraries "/home/me/arduino-1.6.8/libraries" -libraries "/home/me/Arduino/libraries" -fqbn=arduino-beta:samd:arduino_zero_edbg -ide-version=10608 -build-path "/tmp/buildb6ea5fcceab80e8a4f56da35730111a7.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "/home/me/Arduino/libraries/USBHost/examples/KeyboardController/KeyboardController.ino"
"/home/me/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10608 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/cores/arduino" "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero" "/tmp/buildb6ea5fcceab80e8a4f56da35730111a7.tmp/sketch/KeyboardController.ino.cpp" -o "/dev/null"
"/home/me/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10608 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/cores/arduino" "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero" "-I/home/me/Arduino/libraries/USBHost/src" "/tmp/buildb6ea5fcceab80e8a4f56da35730111a7.tmp/sketch/KeyboardController.ino.cpp" -o "/dev/null"
"/home/me/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10608 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/cores/arduino" "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero" "-I/home/me/Arduino/libraries/USBHost/src" "/home/me/Arduino/libraries/USBHost/src/KeyboardController.cpp" -o "/dev/null"
"/home/me/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"  -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10608 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Zero"'     "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/CMSIS/Include/" "-I/home/me/.arduino15/packages/arduino/tools/CMSIS/4.0.0-atmel/Device/ATMEL/"  "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/cores/arduino" "-I/home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero" "-I/home/me/Arduino/libraries/USBHost/src"  [... cut ...]
/home/me/Arduino/libraries/USBHost/src/address.h:290:4: error: 'struct UsbDevice' has no member named 'address'
    if (thePool[i].address)
    ^
Multiple libraries were found for "KeyboardController.h"
 Used: /home/me/Arduino/libraries/USBHost
 Not used: /home/me/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/libraries/USBHost
Using library USBHost at version 1.0.5 in folder: /home/me/Arduino/libraries/USBHost 
exit status 1
Error compiling for board Arduino/Genuino Zero (Programming Port).
```

facchinm, 2016-04-05:
Hi @bbx10 ,
the library examples bug has been posted here https://github.com/arduino/Arduino/issues/4762, quite unfortunately SAM and SAMD examples use different APIs so they won't compile, but at least now the builder chooses the right architecture :smile: