arduino/ArduinoCore-samd #89: attachInterrupt() will not work with sleep mode (known as standby mode for ATSAMD21G18A)

rocketscream, 2016-01-11:
When putting the Arduino Zero to sleep through the standby function call (provided by RTCZero) and using the attachInterrupt() to wake the processor, device will not be able to wake up. The EIC_WAKEUP register needs the pin corresponding WAKEUPENx bit set as mentioned in section 20.8.9 of the SAMD21 datasheet to enable the attachInterrupt to wake the processor during standby.

agdl, 2016-01-11:
@rocketscream yes, the RTC library is intended to be used with functions related to RTC event, so for example an alarm match event

rocketscream, 2016-01-11:
There will be cases where other wake up resources is required on top of the RTC. For example, RTC wake up is used to perform a periodic tasks while an external interrupt (from a sensors, etc) can be used based on condition. I have started to test on adding the WAKEUPEN register configuration and I think it will be beneficial for users. Will make a pull request if you guys don't have any plans to add this important feature.

agdl, 2016-01-11:
feel free to open a PR it will be of course examined and eventually merged!

drewfish, 2016-01-11:
I ran into the same issue and was able to work around it by setting WAKEUPEN myself. It would be nice if the WAKEUPEN change was built into attachInterrupt().

When I adjusted WAKEUPEN myself I was able to get a zero clone (SparkFun SAMD21 Mini Breakout) to use only 1.3mA while sleeping, and wake from either RTC event or from a pin change.

rocketscream, 2016-01-11:
@drewfish On my custom board, I managed to achieve 20 uA of sleep current which includes a PMIC with battery charger, serial flash and also a radio. I will work on the WAKEUPEN later today and test on every single interrupt capable pin.

rocketscream, 2016-01-12:
Made a pull request #90.

sandeepmistry, 2016-02-23:
https://github.com/arduino/ArduinoCore-samd/pull/90 has been merged, so closing this.

closed 2016-02-23 by sandeepmistry

javorosas, 2016-05-06:
I'm using version 1.6.5 in a MKR1000 and still can't make this work. When I push the button it goes to sleep. When I push it again it doesn't seem to trigger the interrupt. Here's my code:

```
#include <RTCZero.h>

const byte button = 5;
bool ledIsOn = false;
volatile bool shouldBeSleeping = false;

RTCZero rtc;

void setup() {
  rtc.begin();
  pinMode(button, INPUT_PULLUP);
  Serial.begin(115200);
  attachInterrupt(button, interr, RISING);
  Serial.println("== Interrupt test ==");
}

void loop() {
  if (shouldBeSleeping) {
    Serial.println("Now going to sleep");
    rtc.standbyMode();
  }
  digitalWrite(LED_BUILTIN, ledIsOn = !ledIsOn);
  delay(1000);
}

void interr() {
  Serial.println("Interruption triggered");
  shouldBeSleeping = !shouldBeSleeping;
}

```

rocketscream, 2016-05-06:
Are you sure the interrupt should be rising when the button pressed? Shouldn't it be falling or low instead as there's pull up which indicate button is active low?

javorosas, 2016-05-06:
You're right. It shouldn't be a pullup. I've just changed it to normal INPUT and it's still not triggering the second interruption. The thing is, if I comment out the `rtc.standbyMode();` line it triggers the interruption every time.

javorosas, 2016-05-06:
I tried with the different interruptions and HIGH and LOW are the only ones that seems to be working. CHANGE, RISING and FALLING won't trigger the interrupt while in sleep mode.

javorosas, 2016-05-06:
I created a new issue for this (#142)