arduino/ArduinoCore-samd #91: Wire.requestFrom() differences AVR / SAMD

sandeepmistry, 2016-01-13:
Moved from: https://github.com/arduino/Arduino/issues/4329

@extrapixel:

> Repeated calls to beginTransmission()/ write() / endTransmission() / requestFrom() behaves differently on AVR and SAM. I think they should either:

> * behave the same or
> * be documented properly

>On UNO:
>Wire.requestFrom(XxXX, n);
> if the received data is shorter than 'n' or not all data is read via Wire.read(), the Wire.available() buffer is empty in the next iteration.

> on ZERO
> Wire.requestFrom(XxXX, n);
> if the received data is shorter than 'n' or not all data is read via Wire.read(), the Wire.available() buffer is not empty, but padded with '0' characters up to 32 bytes and these stay in the buffer even in the next iteration

> (same for IDE 1.6.6 and 1.6.8 hourly 2015/12/21 04:13 )

sandeepmistry, 2016-01-19:
@extrapixel the Uno/AVR behaviour of emptying the buffer on the next ```Wire.requestFrom``` is the desired behaviour. I've submitted #96 to fix this in the SAMD core.

How were you triggering/testing the following?
> on ZERO
> Wire.requestFrom(XxXX, n);
> if the received data is shorter than 'n' or not all data is read via Wire.read(), the Wire.available() buffer
> is not empty, but padded with '0' characters up to 32 bytes and these stay in the buffer even in the next > iteration

From my understanding of the code base and I2C standard, if the I2C master requests ```n``` bytes on a read, ```n``` bytes will be read.

extrapixel, 2016-01-20:
> From my understanding of the code base and I2C standard, if the I2C master requests n bytes on a read, n bytes will be read.

I don't know what is the correct behaviour. All I say is that this behaves differently on Zero and the classic boards (UNO, MEGA).

Use case: a sensor that sends data. The data is max 32 in length. But it can be shorter (depending on the sensor reading / configuration). So always 32 are requested. Then data is read up to the terminating character, can be less than 32 bytes, let's say 25 bytes.. On UNO, after a new call to begin/endTransmission and requestFrom(), the buffer will be empty. On a Zero, there are still the 7 bytes in the buffer.

sandeepmistry, 2016-01-20:
Just to clarify, is this the sequence?

1) ```Wire.requestFrom(address, 32);```
2) ```Wire.available()``` returns 32 (the slave device is forced to return 32 bytes, as part of I2C standard - last 7 bytes will probably not have "valid" data).
3) 25 bytes are read via ```Wire.read()```
4) ```Wire.available()``` is now 7
5) ```Wire.requestFrom(address, 32);``` is called again
6) ```Wire.available()``` returns 32 on the Uno, however 39 is returned on the Zero.

If this is the issue, #96 will change the behaviour in 6) to match the Uno.

cmaglie, 2016-03-30:
Fixed by #96

closed 2016-03-30 by cmaglie