arduino/ArduinoCore-samd #94: digitalWrite() unnecessarily activates the pull-up resistor on the specified pin

Dirk67, 2016-01-19:
digitalWrite() unnecessarily activates the pull-up resistor on the specified pin.

the issue with the pull-up in digitalWrite(), 
(other than the fact that this line shouldn't be there),
is because the pin configuration register (PINCFG) isn't logically ORed with the bit mask when the pull-up bit (PULLEN) is enabled, 
instead the mask is copied directly into the register. 
This effectively clears all the other bits in the register including the input enable (INEN).

--> https://forum.arduino.cc/index.php?topic=367517.msg2534096#msg2534096
--> http://forum.arduino.cc/index.php?topic=334073.msg2306580#msg2306580

spiderkeys, 2016-01-20:
I also commented out that line in digitalWrite(), since our custom board has its own pull-up resistors and it was just unnecessarily pulling down the current on many of the lines.

```c++
  PIO_INPUT,            /* The pin is controlled by PORT and is an input. */
  PIO_INPUT_PULLUP,     /* The pin is controlled by PORT and is an input with internal pull-up resistor enabled. */
  PIO_OUTPUT,           /* The pin is controlled by PORT and is an output. */
```
I think it might be prudent to add one more PIO type here, PIO_OUTPUT_PULLUP, and have the functions in the wiring files respect that, or add a separate property in the pin description structure for resistor configuration altogether.

Nothing critical, but it had me scratching my head for a bit, wondering why my LED was barely turning on.

sandeepmistry, 2016-01-21:
@Dirk67 the goal of enabling the pull-up was to maintain compatibility with AVR core behaviour:

The following is equivalent to ```pinMode(pin, INPUT_PULLUP);``` on AVR:
```arduino
pinMode(pin, INPUT);
digitalWrite(pin, HIGH);
```

However, as you mentioned above, the core is currently clears input enable, so this does not work.

@Dirk67 @spiderkeys I've submitted #101 to resolve this, please review and/or test it out if you have time.

sandeepmistry, 2016-07-08:
Closing since #101 is merged now.

closed 2016-07-08 by sandeepmistry