arduino/ArduinoCore-samd #108: SERCOM + yield() ?

malachib, 2016-02-07:
I can't be sure without more testing but it seems SERCOM blocks indefinitely when I2C is expected but not present; under these circumstances a yield() statement in the while loops and also a timeout seems desirable.

spiderkeys, 2016-02-07:
@malachib We ran into several issues where the I2C interface can lock up, so I redesigned the I2C part of the SERCOM.h/cpp files and the Wire library. After more testing and refinement, I planned to open an issue+PR. In the mean time, you can look at my changes and try them out for yourself:

https://github.com/OpenROV/openrov-arduino-cores/tree/dev/samd

malachib, 2016-02-08:
Nice!  Will do.  I'm playing with this nifty lib [https://github.com/mikaelpatel/Arduino-Scheduler](url) and suddenly people who play nice and yield are very desirable =)

jakorten, 2016-04-22:
I will test both ideas. Having some problems in my SAMD21 based system with freeze-ups, so using a scheduler and solutions to reinit would be very desirable. 

B.t.w. correct links are now:
https://github.com/OpenROV/openrov-arduino-cores/tree/dev/hardware/openrov/samd
https://github.com/mikaelpatel/Arduino-Scheduler