arduino/ArduinoCore-samd #109: String() class fails to construct properly with floats and doubles

mwmckee, 2016-02-10:
Hitting an issue where, on the MKR1000, using Arduino's String class constructor fails to create a string when given an initial value that is a float or a double.

Simple repro: Serial.print(String(1.1)) doesn't print anything.

A repro program is attached. This program works when targeting an Uno. This is with the following two packages installed:
◾Arduino SAMD Boards (32-bits ARM Cortex-M0+), version 1.6.2
◾Arduino SAMD Core - MKR1000 (build 1), version 1.6.2-build-2

[string_bug_repro.txt](https://github.com/arduino/ArduinoCore-samd/files/125370/string_bug_repro.txt)

mwmckee, 2016-02-10:
A workaround is to use the following line of code to force link `printf`'s `float` support

sandeepmistry, 2016-02-11:
@mwmckee I've submitted PR #111 to correct this, @Arduino bot has created a build that you can try out  here: https://github.com/arduino/ArduinoCore-samd/pull/111#issuecomment-183071608.

PR #111 (2016-02-18): Force link _printf_float in dtostrf
Resolves #109.

```_printf_float``` will only get linked in when ```dtostrf``` is used.

Alternative is to add a ```-u _printf_float``` linker flag, but this will increase sketch size for all sketches even if float support for printf is not needed.
(https://github.com/32bitmicro/newlib-nano-1.0/blob/master/newlib/README.nano)
closed 2016-02-18 by cmaglie