arduino/ArduinoCore-samd #115: USBHost calls wiring core init() by accident?

ladyada, 2016-02-14:
hey so this only comes up if you do some modifications to the example code but Init() in the USBHost library calls init()
yet there is no "init()" in the library...there used to be in the original library
https://github.com/arduino-libraries/USBHost/blob/f563445a39da22798bd7985ad3faf857dfacda4e/src/Usb.cpp#L42
but that code is now just in Init.

so instead it ends up calling wiring's init() call, which resets GCLK and some other things. you can change the init() call to add other args to see where the call ends up
```
C:\Users\ladyada\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.2\libraries\USBHost\src\Usb.cpp:40:16: error: too many arguments to function 'void init()'

  init(1,2,3,4,5);

                ^

In file included from C:\Users\ladyada\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.2\cores\arduino/Arduino.h:84:0,

                 from C:\Users\ladyada\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.2\libraries\USBHost\src\Usb.cpp:20:

C:\Users\ladyada\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.2\cores\arduino/wiring.h:25:13: note: declared here

 extern void init(void);
```

fix is just removing the call to init()

bbx10, 2016-02-14:
I found another call to init in Task so I removed that one also.

Removing the calls to init fixes a problem I saw using LED animation and USB keyboard. They work fine separately but together the LEDs did not work. With this fix and other PRs for USB Host (#113, #100, #88), they work fine. The idea is the user enters any text string on the keyboard then presses Enter. The text then scrolls on the LED matrix until another string is entered.

ladyada, 2016-02-15:
thx i changed that init to Init (& removed the uh_init) and that works as well

bbx10, 2016-02-15:
Ok, I tried the same change and it works fine. 

Thanks for fixing this problem. I burned some time chasing this bug but never found the cause.

ladyada, 2016-02-15:
thx for fixing the blocking delay! 
#teamwork :+1:

cmaglie, 2016-02-18:
Based on the discussion here this is the fix I've applied: 

https://github.com/arduino/ArduinoCore-samd/commit/261c375c4f19e32230f9f1bade4bfc9dc8078005

Awesome work, thank you very much for sharing!

commit 261c375c (2016-02-18): USBHost: removed wrong init() calls

USBHost used to have an init() method that now is incorporated in
Init(). The leftover init() calls in the library were calling
the init() function in the variant files instead (that has nothing
to do with the USBHost library).

Removing the calls to init() fix it.

Fix #115
closed 2016-02-18 by cmaglie