arduino/ArduinoCore-samd #120: Zero Bootload Compile Error

NateD4, 2016-03-03:
When compiling with Atmel Studio 7.2 this error occurs:

board_driver_led.h(23,17): error: sam.h: No such file or directory
		 #include <sam.h>
		                 ^
		compilation terminated.

I've tried locating the file manually or including it in the project.

Does this bootloader compile against ASF?

sandeepmistry, 2016-04-04:
@NateD4 have you followed the updated [bootloader readme](https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero#1--prerequisites)?

@aethaniel is there any other special setup required to build the boot loader with Atmel Studio?

MickMad, 2016-04-05:
@sandeepmistry I have the same issue, and I can confirm that I followed the bootloader readme; after all, I just had to open the atsln file, close the ASF Wizard, and click on build solution. I can confirm also that the bootloader with an older version of the repository I had on my PC works fine (it's 46 commits old as of now)

ladyada, 2016-04-11:
hey i did have this happen to me a while ago, i documented the full process here
https://learn.adafruit.com/compiling-m0-atsamd21-bootloader/compile
Make sure you've installed the Arduino SAM and SAMD packages through the board manager!

NateD4, 2016-04-11:
I've installed them with the board manager. There's a glitch in the command line toolchain. Also Arduino is using version 4.0 of the Atmel ASF which appears to be pieced together from the asf-standalone-archive.

This happened to me for some reason it misses the sam.h file. I found if I directly linked to the samd.h file in the source files it seemed to work. It's odd because the make script should scan the sam.h include and header files to get directed to the samd.h header file by the include. But, for some reason it never linked from the sam.h where they include the samd.h files. Of course the samd.h file includes the samd21.h which then includes each of the samd21*.h files for example samd21J18a.h

I was able to get better results by changing the include directory to the asf-standalone. However it may not be necessary.

NateD4, 2016-04-11:
ladyada have you ever altered the D21 bootloader?

ladyada, 2016-04-16:
yah made various changes - my 'golden path' documented above works fine for me (shrug) :)

pocketmoon, 2016-11-05:
Getting the same. Where is sam.h from  ?

Gius-8, 2016-11-22:
Hello,
with @ladyada method and the old Makefile to me it worked well.
I recently updated the core.
When it is compiling the bootloader I get the error:

`
board_driver_led.h:23:17: fatal error: sam.h: No such file or directory
 #include "sam.h"
                 ^
compilation terminated.
make: *** [Makefile:131: build/board_driver_led.o] Error 1
`

How can I fix?

jrsmiley, 2017-01-01:
I had the same problem.  The issue is that the Arduino SAMD core started using the CMSIS-Atmel package as of version 1.6.9.  This moved the location of sam.h and all of its dependencies to a different directory and Atmel Studio can't find it using the built-in template for creating projects from Arduino sketches.

I have not been able to find a way to tell Atmel Studio where the new location is, but I worked around this by creating a Windows "junction" link using the old pathname to point to the new pathname:

```
mkdir C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel\Device
cd C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel\Device
mklink /J ATMEL C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.0.0\CMSIS\Device\ATMEL
```

After that, creating new SAMD projects in Atmel Studio 7 using the Arduino template will compile and debug as expected.

SAMD core 1.6.8 and prior
C:\Users\jrsmi\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel\Device\ATMEL

SAMD core 1.6.9 and later
C:\Users\jrsmi\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.0.0\CMSIS\Device\ATMEL

> From the changelog
```
SAMD CORE 1.6.9 2016.12.12
* MKR1000: Enabled PWM on pin 11, it is shared with pin A4
* SPI: Optimized SPI.transfer(...)
* Added linker flags for the ARM Math library
* Added new I2S library
* Bootloader: Check sketch vectors before waiting for double tap
* Upgraded CMSIS version to 4.5.0, and use new CMSIS-Atmel package
* Added Arduino MKRZero board.
```

Also note that older versions of the Arduino IDE (1.6.8 and prior) will not work with SAMD Core versions 1.6.9 or later due to a similar issue (the old IDE can't find sam.h).

cmaglie, 2017-01-02:
> board_driver_led.h:23:17: fatal error: sam.h: No such file or directory #include "sam.h" ^ compilation terminated. make: *** [Makefile:131: build/board_driver_led.o] Error 1

The file sam.h is part of CMSIS package that is installed with the Arduino IDE Boards Manager when you install the latest samd package.

The `Makefile` provided with the source code should automagically find the CMSIS path (once installed thorugh the Arduino IDE) here:

https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/Makefile#L20

Anyway, you can find the CMSIS package here:

https://downloads.arduino.cc/CMSIS-4.5.0.tar.bz2
https://downloads.arduino.cc/CMSIS-Atmel-1.0.0.tar.bz2

bitron, 2017-01-13:
I just updated the SAM and SAMD cores (1.6.8 -> 1.6.11) using the Boards Manager (Arduino IDE 1.6.12) and that solved the issue in my case.

steamboating, 2017-02-07:
I Did a fresh installation of Arduino IDe 1.6.13 with the new SAM and SAMD Boards plus a new Atmel 7 installation and I still have the problems described above on a Windows 8 System
Frustrating!

facchinm, 2017-02-08:
@steamboating  https://github.com/arduino/ArduinoCore-samd/pull/212 contains a fix for the issue you are encountering. The previous samd core (1.6.11) works flawlessly btw, if you don't want to wait for the next core update. Please note that the paths are correct only if you open the project from `C:\Users\yourname\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.xx\bootloaders\zero\Makefile`

facchinm, 2017-02-08:
Solved by merging #212

closed 2017-02-08 by facchinm

steamboating, 2017-02-08:
Thank you facchinm for the quick reply!
This #212 helped me to get rid of the <sam.h> problem but now I went into the above described libsam issue - will try to use 1.6.11