arduino/ArduinoCore-samd #127: USB Host Hub Support is Unclear

steve-chamberlin, 2016-04-01:
The SAMD library description specifically says "USBHost does not support devices that are connected through USB hubs", however the library contains a usbhub.cpp and .h files, and the USB_desc example appears to work with hubs. Is the library description out of date, or is the hub implementation incomplete maybe? I don't yet have SAMD hardware to try this on, and want to confirm whether I could host a keyboard and a mouse simultaneously before I buy the hardware.

sandeepmistry, 2016-04-04:
@agdl would you be able to look into this and provide an answer?

sandeepmistry, 2016-07-11:
@bbx10 @jcbuda do you have any information on this?

I tried using the [USB_desc.ino](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/USBHost/examples/USB_desc/USB_desc.ino) example with a powered hub, but it would only print the info. of the hub, not the devices connected it it.

I also tried adding a `USBHub  Hub1(&usb);` line like was indicated in the sketch, but then nothing was printed out.

bbx10, 2016-07-11:
The last time I looked, there is no hub support.