arduino/ArduinoCore-samd #133: Wire library will not compile if transmitting a zero to bus

dlabun, 2016-04-13:
If you attempt to send a zero over the I2C bus the code will fail to compile. This error appears to affect all versions of the SAMD core but does not affect the AVR or SAM platforms. I ran into it when trying to program the time on an RTC chip.

Using the following very simple code will induce the error:

```
#include <Wire.h>

void setup() {
  // put your setup code here, to run once:

 Wire.begin();
 Wire.beginTransmission(B1101111);
 Wire.write(0x00);
 Wire.endTransmission();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

This will cause an error of: 

```
error: call of overloaded 'write(int)' is ambiguous

  Wire.write(0x00);

                 ^

C:\Users\Douglas\Documents\Arduino\sketch_apr13b\sketch_apr13b.ino:8:17: note: candidates are:

In file included from C:\Users\Douglas\Documents\Arduino\sketch_apr13b\sketch_apr13b.ino:1:0:

C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\libraries\Wire/Wire.h:50:12: note: virtual size_t TwoWire::write(uint8_t)

     size_t write(uint8_t data);

            ^

In file included from C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\cores\arduino/Stream.h:26:0,

                 from C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\cores\arduino/HardwareSerial.h:24,

                 from C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\cores\arduino/Arduino.h:73,

                 from C:\Users\Douglas\AppData\Local\Temp\build2263b14e086498ed09b50d77f76f28f0.tmp\sketch\sketch_apr13b.ino.cpp:1:

C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\cores\arduino/Print.h:48:12: note: size_t Print::write(const char*)

     size_t write(const char *str) {

            ^

Using library Wire at version 1.0 in folder: C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.5\libraries\Wire 
exit status 1
call of overloaded 'write(int)' is ambiguous

```

sandeepmistry, 2016-04-13:
Hi @dlabun,

I've submitted #134 to resolve to this, there are some instructions to try it out in https://github.com/arduino/ArduinoCore-samd/pull/134#issuecomment-209574815

dlabun, 2016-04-14:
The patch looks good, I2C bus is working as expected.

sandeepmistry, 2016-04-14:
@dlabun great, thank you for testing it out!

cmaglie, 2016-04-19:
Fixed by #134

closed 2016-04-19 by cmaglie