arduino/ArduinoCore-samd #136: Interrupts Feather M0 from Adafruit

hellbound1980, 2016-04-18:
Working with a Feather M0 from Adafruit and I was unable to get interrupts functioning properly.

For instance....when using A1 through A5 as interrupts, a change on A5 would cause the interrupt routine for A1 to be run. Meanwhile, a change on A1 through A4 would do absolutely nothing.

Did not take the time to test further due to the time sensitive nature of my project, but I needed to do some interrupts on pins 5 and 6 and was unable to get those functioning either.

Not sure if this effects the Arduino Zero as well, but I imagine it does in some way.

sandeepmistry, 2016-04-18:
I've tried the following sketch on the Zero:
```arduino
void setup() {
  Serial.begin(9600);
  while(!Serial);

  Serial.println("hello");

  pinMode(A0, INPUT_PULLUP);
  pinMode(A1, INPUT_PULLUP);
  pinMode(A2, INPUT_PULLUP);
  pinMode(A3, INPUT_PULLUP);
  pinMode(A4, INPUT_PULLUP);
  pinMode(A5, INPUT_PULLUP);

  attachInterrupt(A0, onA0Change, CHANGE);
  attachInterrupt(A1, onA1Change, CHANGE);
  attachInterrupt(A2, onA2Change, CHANGE);
  attachInterrupt(A3, onA3Change, CHANGE);
  attachInterrupt(A4, onA4Change, CHANGE);
  attachInterrupt(A5, onA5Change, CHANGE);
}

void loop() {
}

void onA0Change() {
  Serial.println("A0 change");
}

void onA1Change() {
  Serial.println("A1 change");
}

void onA2Change() {
  Serial.println("A2 change");
}

void onA3Change() {
  Serial.println("A3 change");
}

void onA4Change() {
  Serial.println("A4 change");
}

void onA5Change() {
  Serial.println("A5 change");
}
```

Pins A1 to A4 are ok, but A0 and A5 have issues. ```onA5Change``` is called when I change the digital level on either A0 or A5.

Based on the [variant.cpp](https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.cpp#L141-L146) both A0 and A5 share ```EXTERNAL_INT_2```.

cc/ @ladyada

sandeepmistry, 2016-04-18:
@cmaglie I'm not sure if this is by design, but the following pins share external interrupts in the Zero variant:

* ```EXTERNAL_INT_2```: A0, A5, 10
* ```EXTERNAL_INT_4```: A3, 6
* ```EXTERNAL_INT_5```: A4, 7
* ```EXTERNAL_INT_6```: 8, SDA
* ```EXTERNAL_INT_7```: 9, SCL
* ```EXTERNAL_INT_9```: A2, 3
* ```EXTERNAL_INT_10```: TX, MOSI
* ```EXTERNAL_INT_11```: RX, SCK

hellbound1980, 2016-04-18:
What about if you use a discrete value (LOW or HIGH) instead of a change? 
My specific application I was using a LOW value (as I didn't want to deal with a value change interrupt).
I'm not sure if that will generate something different, but using the current version of the Adafruit inclusion did not have the same effects.

In particular, I know for a fact that a LOW value on A5 was causing the subroutine for A1 (my application only used A1 through A5, not A0 since I was aware of the fact that A0 and A5 shared the same interrupt).

I will have to attempt to try some new test code tonight on A1 through A5 as I still have it wired up but have been using software polling instead to progress my project.

sandeepmistry, 2016-04-18:
No issues with using LOW or HIGH instead of change on the Zero.

On the Arduino side, please note as mentioned in the [attachInterrupt documentation](http://www.arduino.cc/en/Reference/AttachInterrupt) not all pins have interrupt capabilities, and some may share the same external interrupt.

I'm going to close this, as all is good on the Arduino Zero side. I suggest you follow up with Adafruit on this: https://www.adafruit.com/support

closed 2016-04-18 by sandeepmistry

drewfish, 2016-04-20:
(Not sure if it's related, but my understanding was that it's suggested not to `Serial.println()` in an interrupt.)

drewfish, 2016-04-20:
I wrote a library to dump the SAMD configuration registers, which might be of some help in debugging.
https://github.com/drewfish/arduino-ZeroRegs

Adding this at the bottom of setup() might be helpful:
```c++
    ZeroRegOptions opts = { Serial, false };
    printZeroRegsEIC(opts);
```

ladyada, 2016-07-13:
heya this was on an old ToDo list and i just did a look with the above test code that sandeep posted. A1 thru A4 work great.

But basically it looks like you can use *either* A0 or A5 (just verified with the Feather M0 SAMD which is really just a light variation of the official SAMD)
however, you cannot use both because the interrupt # is the same but
```
   // Assign pin to EIC 
  pinPeripheral(pin, PIO_EXTINT);
```
gets called twice - once for each pin, and thats likely causing confusion in the core (well thats what I think is the problem)

so yeah, just use one or the other! :)

hellbound1980, 2016-07-13:
Thank you!

I finished my project using another method. My light saber works great, and
I absolutely loved the feather m0. It is very powerful.

Thank you Ladyada!

Aaron Lunger