arduino/ArduinoCore-samd #140: Serial1.flush() blocks if called after Serial1.begin() and before port is written to.

nsted, 2016-04-27:
Caused problems while working on RS485 project on Arduino Zero. I flushed the port when switching between RX and TX mode, including at startup (ie before data is sent), which caused the program to block. 

----

Test:
```
void setup() {
  Serial1.begin(9600); 
  SerialUSB.begin(9600);

//  Serial1.write(1);           // uncomment to prevent blocking

  SerialUSB.flush();            // doesn't block
  Serial1.flush();              // blocks everything below if called after Serial1.begin() and before any data sent.

  delay(2000);
  Serial1.println("Serial1 up and running");  
  SerialUSB.println("SerialUSB up and running");
  delay(3000);
}

int i;

void loop() {
  // put your main code here, to run repeatedly:
  Serial1.println(i++);
  SerialUSB.println(i);
}
```

The issue appears to be tied to this method:
```
void SERCOM::flushUART()
{
  // Wait for transmission to complete
  while(!sercom->USART.INTFLAG.bit.TXC);
}
```

----

rocketscream, 2016-05-04:
The only way to avoid this would be checking the data register empty bit before the while loop:

```
void SERCOM::flushUART()
{
  if(isDataRegisterEmptyUART()) return;
  // Wait for transmission to complete
  while(!sercom->USART.INTFLAG.bit.TXC);
}
```

sandeepmistry, 2016-05-27:
I agreed with @rocketscream's suggestion above.

@rocketscream would you be interested in submitting a pull request?

rocketscream, 2016-05-30:
@sandeepmistry Okay, did a pull request #144

sandeepmistry, 2016-05-30:
Closed via @rocketscream's change in https://github.com/arduino/ArduinoCore-samd/commit/bfd2a0bb28f7668b2b24a1ec72bf8c1e1d5cbd2e.

closed 2016-05-30 by sandeepmistry