arduino/ArduinoCore-samd #147: SPI library does not limit the maximum SPI clock frequency as per the datasheet

joseangeljimenez, 2016-06-05:
While the current SPI.h tries to safe-limit the maximum SPI clock frequency of the SAMD core, it does so in a weird way, assuming F_CPU to be always 48000000L and only for the SAMD21G18A part.
https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SPI/SPI.h

This has two problems,

- Will fail for the SAMD21G18A if F_CPU takes a different value.
- Will fail for any other SAMD21 parts, as the code relaxes lowers in this case the value of SPI_MIN_CLOCK_DIVIDER to 2.

A comment in the code also states that the maximum SPI clock frequency is not specified in the datasheet. However, it is. According to the datasheet for the SAM D21E, D21G and SAM D21J ("Atmel-42181I-SAM L21_Datasheet_Complete-03/2016"), page 997,

- tSCK SCK period in master mode is typically 84ns, which roughly translates into 11.9MHz.
- According to Atmel, the value above is based on simulation. It is not tested in production.

I have prepared a patch for correcting SPI.h so that it effectively limits the max SPI clock frequency to 12000000L, for all F_CPU values and all SAMD21 parts. Is it OK if I submit a pull request?

sandeepmistry, 2016-06-06:
Hi @joseangeljimenez,

Yes, please go ahead an submit a pull request for this change.

joseangeljimenez, 2018-01-09:
https://github.com/arduino/ArduinoCore-samd/pull/292

dhalbert, 2018-01-30:
Based on testing, a 24 MHz SPI can work, though it may be out of spec. Should Arduino limit the value to 12 Mhz, per #292, or should it simply be documented as "may not work"? Some discussion with further URL's in https://github.com/adafruit/circuitpython/pull/560.

joseangeljimenez, 2018-01-31:
Hi @dhalbert et al, let me clarify this a little bit more,

- 24 MHz should work absolutely fine (at least inside the MCU) and should not be considered overclocking. The SAMD21 datasheet specifies a typical SPI SCK period (tSCK) of 42 ns, which translates into a typical SPI clock of 23.8 MHz. Hence, a 24 MHz clock is within a 1% tolerance with respect to the typical value given in the datasheet.

- Despite the previous, in my pull request https://github.com/arduino/ArduinoCore-samd/pull/292 I conservatively kept 12 MHz as the maximum clock for the SPI as this was the historical behavior of this code in the ArduinoCore repo.

- I guess the historical 12 MHz top is due to the potential issue that enthusiasts prototyping with "air wires" and breadboards, without a proper PCB and layout, will get a malfunctioning SPI bus at higher frequencies. But that is due to quick & dirty prototyping techniques not due to an issue with the SAMD chips.

- My main goal for the above pull request is that the SPI divider is calculated in the same way no matter which exact SAMD variant is used.

I hope this helps to clarify the issue and my motivations behing the pull request.

modmatic, 2020-02-06:
@joseangeljimenez Could you clarify the calculation on your PR for SAMD?

It seems that `SPI_MIN_CLOCK_DIVIDER` will have a value of 5 for, say, a 48MHz `F_CPU`, which results in a **9.6MHz** clock instead of 12MHz. See details below:

In `SPISettings::init_AlwaysInline()`, where the SPI clock frequency is being set based on the desired `clock` value:
```
this->clockFreq = (clock >= (F_CPU / SPI_MIN_CLOCK_DIVIDER) ? F_CPU / SPI_MIN_CLOCK_DIVIDER : clock);
```
This will result in `clockFreq` being `48,000,000 / 5 = 9,600,000` when `clock = 12,000,000`.

When the settings are finally applied to the baud rate register (after calling `SPIClass::beginTransaction()`), the resulting baud rate is also `9.6MHz:
```
sercom->SPI.BAUD.reg = calculateBaudrateSynchronous(baudrate);  // baudrate == settings.clockFreq == 9,600,000 here
``` 
which results in:
```
sercom->SPI.BAUD.reg = SERCOM_FREQ_REF / (2 * baudrate) - 1;  // calculation also in SAMD21 data sheet
```
which results in 1.5 if `SERCOM_FREQ_REF = 48,000,000`.

Per the data sheet, the resulting baud can be calculated as follows (essentially, solving the equation above for `baudrate`):
```
baudrate = SERCOM_FREQ_REF / (2 * (sercom->SPI.BAUD.reg + 1))
```
which results in 9,600,000

> SIDE NOTE:
> Shouldn't `SERCOM_FREQ_REF` be used when computing `SPI_MIN_CLOCK_DIVIDER` rather than `F_CPU`?