arduino/ArduinoCore-samd #150: Upload using programmer - Atmel ICE is not working.

FRVisser, 2016-06-27:
Hi,

The option "upload using programmer" using the Atmel ICE is not working. I selected the Atmel SAM-ICE programmer. The burning of the bootloader works fine though. 

  ``` shell
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-19:16)
  Licensed under GNU GPL v2
  For bug reports, read http://openocd.org/doc/doxygen/bugs.html
  debug_level: 2
  Info : only one transport option; autoselect 'swd'
  adapter speed: 500 kHz
  adapter_nsrst_delay: 100
  cortex_m reset_config sysresetreq
  Info : CMSIS-DAP: SWD  Supported
  Info : CMSIS-DAP: JTAG Supported
  Info : CMSIS-DAP: Interface Initialised (SWD)
  Info : CMSIS-DAP: FW Version = 01.1C.0067
  Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
  Info : CMSIS-DAP: Interface ready
  Info : clock speed 500 kHz
  Info : SWD IDCODE 0x0bc11477
  Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
  Error: at91samd21g18.cpu -- clearing lockup after double fault
  Polling target at91samd21g18.cpu failed, trying to reexamine
  Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
  target state: halted
  target halted due to debug-request, current mode: Thread 
  xPSR: 0x61000000 pc: 0xfffffffe msp: 0xfffffffc
  ** Programming Started **
  auto erase enabled
  Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
  Error: SAMD: NVM lock error
  Error: Failed to erase row containing 00000000
  Error: error writing to flash at address 0x00000000 at offset 0x00000000
  Runtime Error: embedded:startup.tcl:454: ** Programming Failed **
  in procedure 'program' 
  in procedure 'program_error' called at file "embedded:startup.tcl", line 510
  at file "embedded:startup.tcl", line 454
  the selected serial port at file "embedded:startup.tcl"
```

I just use a empty sketch for testing:

``` arduino
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

gh-megabit, 2016-07-08:
I've seen this too. Note that you can upload with the Atmel ICE by selecting the debug port option from the boards list and uploading normally.

999Count, 2016-07-29:
Hello FRVisser, read your post can you tell me for bootloading burning you used that option. I want to bootload MKR1000 can you help me.

FRVisser, 2016-08-02:
@gh-megabit I can confirm this is working. Only confusing is that the Arduino IDE does not see the atmel-ice, but if you upload it finds the connected interface. It's a bit of a user interface thingy 😃.  Thanks 👍 

@999Count for the Arduino Zero I select the board Arduino/Genuino Zero (Programming Port), and Programmer: "Atmel SAM-ICE". Then press Burn Bootloader. (tip: if its not working, try turning the connector 180 degrees.) Should be the same for MKR1000

closed 2016-08-26 by FRVisser