arduino/ArduinoCore-samd #157: Internal OSC32K Support

mirage335, 2016-07-28:
Have been attempting to find the cause of an inability to use ATSAMD21**_J_**18 with the Arduino IDE. Best guess is lack of support for the lack of an external 32kHz oscillator in my circuits, where space and cost can be at an extreme premium.

Any help would be greatly appreciated, and I will try to keep working on this myself as well. Please see the related github repository for schematics, an extensive history of debugging attempts, working LED blink code from Atmel Studio, example Arduino sketches, and more related documentation.

https://github.com/mirage335/ATSAMD21J18_ArduinoSupport

spiderkeys, 2016-07-29:
You will need to modify which clock is used to drive GCLK1.

Look at: https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/startup.c

You can change it to use the one of the SAMD21J18A's internal oscillators:

32.768kHz internal oscillator (OSC32K)
32KHz ultra-low-power internal oscillator (OSCULP32K)
8MHz high-accuracy internal oscillator (OSC8M)

See page 13 here: http://www.atmel.com/Images/Atmel-42181-SAM-D21_Datasheet.pdf

I believe there should be no problem setting it to OSC32K, but it will not be as stable as an external crystal, and you will need to double check the clock sections of the datasheet to make sure you set up the subsequent clocks that are driven by it correctly (all of this is done in startup.c).

mirage335, 2016-07-29:
Thanks. FWIW, OSC32K is referred to as a "High-Accuracy Internal Crystal Oscillator", which I think will be good enough for many uses.  Right now, I am stuck on porting the existing code that configures XOSC32K. Documentation is lacking at times. Specifically things like this.

`SYSCTRL->XOSC32K.reg = SYSCTRL_OSC32K_STARTUP( 0x6u ) | /* cf table 15.10 of product datasheet in chapter 15.8.6 */`

Chapter 15.8.6 does not seem to exist in the current version of the manual. Consequently, the configuration options in this bitmask are not obvious. Correlating the "sysctrl.h" macros intended to set and verify registers enumerated in the datasheet could be quite time consuming.

If any documentation outside the datasheet was used to build startup.c and board_init.c, that would be very helpful.

mirage335, 2016-07-29:
Based on the datasheet information, I substituted all references to XOSC32K for OSC32K, and removed the XTALEN and EN32K bits relevant to external oscillator information. Bootloader now hangs waiting for oscillator startup.

https://github.com/mirage335/ArduinoCore-samd/commit/040a7b7bfe11be8bff71b149b8f111a8fba259ef

spiderkeys, 2016-07-29:
> mirage335@040a7b7

Sorry, not sure what to suggest concerning the bootloader. I flash my
program directly over SWD and don't use the bootloader, so I am unfamiliar
with how to configure it.

mirage335, 2016-07-29:
Very interesting. The code used for configuring clocks on the bootloader is very similar to the code used for configuring clocks in Arduino sketches, but nothing alike any Atmel Studio examples I can find.

Did you need to do anything special to upload Arduino sketches directly over SWD, or did you just upload the hex file from a temporary directory created by a default Arduino IDE installation?

facchinm, 2016-07-29:
Hi @mirage335 , you can upload a sketch via swd using any programmer supported by openocd if you tweak `programmers.txt` file in samd core. 
We currently support EDBG, AtmeliCE and SAM-iCE as programmers. 
To directly upload a sketch, press the `Upload ` button while pressing `shift`; the caption will change into `Upload using programmer`. If you are planning to avoid using a bootloader you should also change the `ldscript` rule for your board (similar to https://github.com/arduino/ArduinoCore-samd/blob/master/boards.txt#L36 ) to use `flash_without_bootloader.ld` linker script

spiderkeys, 2016-07-29:
@mirage335 In our product, we combine the Raspberry Pi 3 with the SAMD21J18A (and potentially the SAML21) on a custom controller board. We don't have the Ateml EDBG chip and the ICE programmers would be too expensive/bulky, so we use the BCM2835 driver for OpenOCD to program directly over SWD on three of the RPi's GPIO pins (SWDIO, SWCLK, and RESET). We've also gotten this to work from a Beaglebone Black using OpenOCD's sysfsgpio driver, although it is much slower. If you don't have a supported programmer laying around, and you can get the EDBG chip out of the way, this is a cheap and relatively easy way to spin up a SWD programmer.

You can look at our build script for OpenOCD here that enables those features:
https://github.com/OpenROV-Dev/openrov-openocd

And here is the modified OpenOCD cfg file we use for the RPI3 (RPI2 has an official config file in OpenOCD if you want to use it instead):
https://github.com/OpenROV/openrov-cockpit/blob/master/src/system-plugins/platform-manager/platforms/raspberrypi/install/files/openocd.cfg

And lastly, here is the command to flash a resulting .bin file with the above :
```bash
openocd -f /path/to/rpi/openocd.cfg -c "program /path/to/your/sketch.bin; reset; exit"
```

mirage335, 2016-07-29:
Good to know. Still, I have been trying to work on the bootloader code, because it is easy to debug at the moment, and uses the same structure as the sketch code I need to fix.

My boot_init.c code currently does not work even with real Arduino hardware, crashing during Serial or USB init. Consequently, it seems likely the clocks are still not correctly configured.

Have also attempted to direct the clock to an external pin for diagnosis, unsuccessfully.
`
  //Output clock .
  PORT->Group[0].PINCFG[PIN_PA14].bit.PMUXEN = 1;
  PORT->Group[0].PMUX[PIN_PA14/2].reg &= 0xF0;
  PORT->Group[0].PMUX[PIN_PA14/2].reg |= 0x07;
`

mirage335, 2016-08-14:
Much thanks to Kliment for generously solving this problem. His solution is documented at his fork of the GitHub repository.
https://github.com/kliment/ArduinoCore-samd

Also, thanks to everyone generally for providing helpful information.

Now, adding this CrystalFree configuration to the mainline repository might help others use similar barebones setups.

casundra, 2016-08-18:
That is incredibly useful!  I had just started to delve into it, changed clock configs in startup.c, and then when I noticed that just the bootloader hung, figured that it must also use the external crystal.  Thank you Kliment, and thank you for linking this @mirage335 .

tinkerBOY-git, 2019-12-11:
I just tried the bootloader at https://github.com/kliment/ArduinoCore-samd/blob/master/bootloaders/zero/samd21_sam_ba.bin to flash a custom board i made with atsamd21g18a without crystal and this [Wemos Mini board](https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=wemos+samdi21+mini&_sacat=0). I used Atmel Studio 7 and the bootloader uploads just fine.  I can also upload Arduino sketches just fine but uploaded sketches won't work unless i put in a external 32khz crystal. The wemos board has an external crystal and uploaded sketches works fine with it but it won't if I removed the crystal.

Do I have to re-compile the bootloader in order to use the atsamd21g18a without an external crystal?

mirage335, 2019-12-12:
Possibly. Having spent a long time developing Ubiquitous Bash partly for the purpose of containing applications like Arduino, it has been a while since I tested this specifically. It will still be a while longer before I have my development environment ready to do so.

Any test results welcome!

facchinm, 2019-12-16:
@tinkerBOY-git for applications without the 32kHz crystal you need to define `CRYSTALLESS` in both bootloader and core (you can take Nano 33 IoT as reference)

closed 2019-12-16 by facchinm

tinkerBOY-git, 2019-12-17:
> @tinkerBOY-git for applications without the 32kHz crystal you need to define `CRYSTALLESS` in both bootloader and core (you can take Nano 33 IoT as reference)

Yes I have already done that. It looks like it has something to do with the bossac.exe not supporting this particular chip atmsamd21e17d.

mirage335, 2019-12-20:
Then you might be able to use ATMEL Studio to upload the binary? Not ideal for prototyping/debugging, but might be useful for production.