arduino/ArduinoCore-samd #161: No sleep with external interrupt.

FRVisser, 2016-08-26:
We use deep sleep to use the SAMD21 on battery. This is working great with figures around 3-5uA. But with an external interrupt the SAMD21 is still Using 10mA (this is with nothing connected to the interrupt pin). Internal interrupt is working fine, as we use with RTCZero alarm.

sandeepmistry, 2016-08-31:
Hi @FRVisser,

This is a duplicate of https://github.com/arduino/ArduinoCore-samd/issues/142.

Since the EIC is not using the XOSC32K as it's clock source it is inactive in sleep mode.

I'm closing this as a duplicate of #142, please follow that issue for updates. Please see https://github.com/arduino/ArduinoCore-samd/issues/142#issuecomment-223061918 for a work around for now.

closed 2016-08-31 by sandeepmistry

FRVisser, 2016-09-01:
Ah, thanks! I found some help at [forum.sodaq.com](http://forum.sodaq.com/t/using-xbee-to-control-sleep-cycle/456/5)