arduino/ArduinoCore-samd #162: SPI comms issues with SAMD Boards 1.0.9 and above

PaulZC, 2016-09-13:
Hi Gang,

I wrote some code for the Feather Adalogger M0 earlier this year which - amongst other things - reads the temperature from a MAX31855 thermocouple board via SPI and logs it to SD card. I tried to reuse the code again today and it failed. After a lot of head scratching I found that if I regressed my Adafruit SAMD Boards back to 1.0.8 it worked. If I moved on to 1.0.9 or above, it failed... The SPI temperature read falls over and returns zeros. I'm still working on condensing some code to (concisely) illustrate the problem but - at this stage - just thought I would ask if anyone has had similar problems or has a fix? I'm kind of hoping @sandeepmistry or @agdl can shed some light on this? The strange thing is that the SD card write still works - it is just the MAX31855 comms that fall over... Weird - especially since they share the same bus!

I've tried the Adafruit MAX31855 library as well as the Sparkfun version and they both throw up the same problem. It seems to be something fundamental to SPI itself?

Many thanks in advance,
Paul

facchinm, 2016-09-16:
Hi Paul,
the 1.0.9 version looks quite ancient (current is 1.0.13). Can you also reproduce in the latest? However, the right repo for this issue is Adafruit's fork (https://github.com/adafruit/ArduinoCore-samd) unless you are able to reproduce the behaviour with an the official core on an Arduino/Genuino board; I'd move the issue myself if the fork had an issue tracker, so I'm leaving it here for @ladyada

PaulZC, 2016-09-19:
Thanks Facchinm - I should indeed have raised this on the Adafruit fork. My bad! I'll finish condensing the code and then raise it over there.

closed 2016-09-19 by PaulZC

ladyada, 2016-09-23:
i haven't seen any issues raised but note that the MAX31855 likes best to be on its own two pins (no bus sharing), dont have it share the SPI for the SD card!