arduino/ArduinoCore-samd #164: issue with uploading sketch via programmer (sam-ice)

tuxedo0801, 2016-09-21:
Hi,
I have an Elecrow Crowduino M0 SD and have some issues with uploading a sketch.

Environment:
* Arduino IDE 1.6.11 from arduino.cc
* Arduino SAMD Boards (32-bits ARM Cortex-M0+) 1.6.6 via Boardmanager
* Blink Sketch

I've flashed the bootloader with arduino IDE + programmer "ATMEL SAM-ICE". Works great:

```
C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino/bin/openocd.exe -d2 -s C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino/share/openocd/scripts/ -f C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.6/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; init; halt; at91samd bootloader 0; program {{C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.6/bootloaders/zero/samd21_sam_ba.bin}} verify reset; shutdown 
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-22:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.21.0076
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x000041b6 msp: 0x20007fe8
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x000005e8 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 16384 bytes from file C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.6/bootloaders/zero/samd21_sam_ba.bin in 1.898500s (8.428 KiB/s)
** Programming Finished **
** Verify Started **
verified 5972 bytes in 0.521500s (11.183 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

Uploading a sketch via USB works also:

```
Der Sketch verwendet 8.356 Bytes (3%) des Programmspeicherplatzes. Das Maximum sind 262.144 Bytes.
Erzwinge Reset durch öffnen/schließen mit 1200 bps auf dem Port COM12
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
PORTS {COM12, } / {COM12, } => {}
Uploading using selected port: COM12
C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM12 -U true -i -e -w -v C:\Users\<user>\AppData\Local\Temp\buildbb61b02d674ae2bfb0845dfb247346a1.tmp/Blink.ino.bin -R 
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v2.0 [Arduino:XYZ] May  9 2016 16:06:46
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Chip ID      : 10010005
version()=v2.0 [Arduino:XYZ] May  9 2016 16:06:46
Version      : v2.0 [Arduino:XYZ] May  9 2016 16:06:46
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.807 seconds

Write 8508 bytes to flash (133 pages)
write(addr=0x20005000,size=0x1000)
writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)

[==============                ] 48% (64/133 pages)write(addr=0x20005000,size=0x1000)
writeBuffer(scr_addr=0x20005000, dst_addr=0x3000, size=0x1000)

[============================  ] 96% (128/133 pages)write(addr=0x20005000,size=0x140)
writeBuffer(scr_addr=0x20005000, dst_addr=0x4000, size=0x140)

[==============================] 100% (133/133 pages)
done in 0.060 seconds

Verify 8508 bytes of flash with checksum.
checksumBuffer(start_addr=0x2000, size=0x1000) = 917e
checksumBuffer(start_addr=0x3000, size=0x1000) = 9f85
checksumBuffer(start_addr=0x4000, size=0x13c) = 67f9
Verify successful
done in 0.013 seconds
CPU reset.
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
writeWord(addr=0xe000ed0c,value=0x5fa0004)
```

But uploading the same sketch via programmer (again atmel sam-ice):

```
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-22:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.21.0076
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x41000000 pc: 0x000005e8 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 32768 bytes from file C:\Users\<user>\AppData\Local\Temp\buildbb61b02d674ae2bfb0845dfb247346a1.tmp/Blink.ino.elf in 3.812762s (8.393 KiB/s)
** Programming Finished **
** Verify Started **
verified 16700 bytes in 1.427786s (11.422 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

Seems to work because no error is shown. But Arduino is not executing blink sketch and is not accepting any new sketch via USB (even with double-tap-reset). I have to flash the bootloader again (with sam-ice) to get the arduino back into a working state (to accept sketch upload via USB)

What's wrong? Any hints?

br,
Alex

tuxedo0801, 2016-09-22:
Retested with Arduino IDE 1.6.12... Same result.
Retested with "Sloeber Arduino Eclipse IDE" (http://eclipse.baeyens.it/) which uses same Arduino Core Samd 1.6.6  ... Same result as with Arduino IDE.

So it's not the IDE itself, but rather the Core Samd 1.6.6

Went back to 1.6.5 and 1.6.4 with Arduino IDE ... But this gave me some errors on execution. 

A friend of mine is using the same setup with Arduino IDE 1.6.10 and Core Samd 1.6.4, also on Windows.. with success.

No idea what wents wrong.

facchinm, 2016-09-22:
Hi @tuxedo0801 ,
I believe your problem can be related with linker script offsets. 
By default, both programming via USB or via programmer use `flash_with_bootloader.ld`.
This loads the sketch at 0x2000, and the bootloader is responsible to start the sketch from that position. Unfortunately, the bootloader gets erased during the second programming due to its size. 
A patch to address it is here https://github.com/arduino/OpenOCD/pull/2, I'll attach here a binary for Windows so you can test it on your environment.

tuxedo0801, 2016-09-22:
If this is a bug: Question is why a friend of mine is able to upload sketch via atmel ice... 

only difference: he's using a differen arduino zero board.

The board I use: Elecrow Crowduino M0 SD
The board he's using: Arduino M0 Pro (with atmel ice, instead of onboard debugger)

But used samd is exactly the same...

facchinm, 2016-09-22:
M0 pro has 0x4000 as bootloader offset, while our implementation is smaller (thus it gives you more space for sketches) but it encounters the openocd bug. 
The attached file should be ok (the build procedure is not yet validated but it should work)
[openocd.zip](https://github.com/arduino/ArduinoCore-samd/files/487569/openocd.zip)

tuxedo0801, 2017-03-29:
Sorry for late response ...Now using Linux, attached openocd is windows version :-(

But I tried the current "nightlybuild" of samd-core (with linux). Normal "uploading sketch" uses openocd to upload the sketch:

```
/home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5-static/bin/openocd -d2 -s /home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5-static/share/openocd/scripts/ -f /home/achristian/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{/tmp/arduino_build_282547/M0dularisPlusBlink.ino.bin}} verify reset 0x2000; shutdown 
Open On-Chip Debugger 0.9.0-g688b0b6 (2017-01-12-14:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.21.0076
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
Warn : Adding extra erase range, 00000000 to 0x00001fff
wrote 24576 bytes from file /tmp/arduino_build_282547/M0dularisPlusBlink.ino.bin in 2.732562s (8.783 KiB/s)
** Programming Finished **
** Verify Started **
verified 9712 bytes in 0.817418s (11.603 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

Works. Blink sketch runs fine. 
"Uploading via Programmer" gives me:

```
/home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5-static/bin/openocd -d2 -s /home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5-static/share/openocd/scripts/ -f /home/achristian/.arduino15/packages/arduino-beta/hardware/samd/9.9.9-Hourly/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{/tmp/arduino_build_282547/M0dularisPlusBlink.ino.elf}} verify reset; shutdown 
Open On-Chip Debugger 0.9.0-g688b0b6 (2017-01-12-14:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.21.0076
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 32768 bytes from file /tmp/arduino_build_282547/M0dularisPlusBlink.ino.elf in 3.694437s (8.662 KiB/s)
** Programming Finished **
** Verify Started **
verified 17904 bytes in 1.517887s (11.519 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```
Does not work: Blink sketch isn't running.

The difference seems to be the missing "verify reset **0x2000**; shutdown " when using the menuitem "upload via programmer" ...

facchinm, 2017-03-29:
There are a couple of problems here:
- 0.9.0-arduino5 does not contain the right patch, so the bootloader gets erased nevertheless. I'm attaching  0.9.0-arduino6 compiled for Linux which should solve the issue forever.

The "upload using programmer" uses 0 as offset, as you noticed. If you don't change the linker script too, the sketch won't start since it will be flashed in the wrong location.

To allow a clean, bootloader-less, programming, you should change `boardname.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld` into `boardname.build.ldscript=linker_scripts/gcc/flash_without_bootloader.ld` in the core's `boards.txt`
[openocd-0.9.0-arduino6-static-x86_64-linux-gnu.tar.bz2.zip](https://github.com/arduino/ArduinoCore-samd/files/879626/openocd-0.9.0-arduino6-static-x86_64-linux-gnu.tar.bz2.zip)

tuxedo0801, 2018-04-06:
* *ping* *

tuxedo0801, 2018-04-23:
*ping again*

mirage335, 2018-05-26:
Upstream OpenOCD seems to work.
https://sourceforge.net/p/openocd/code/ci/master/tree/

Perhaps now would be a good time to update Arduino's OpenOCD.
https://github.com/arduino/OpenOCD/issues/5

tuxedo0801, 2019-02-23:
*ping* again... Issue still present with 1.6.20 :-(

mirage335, 2019-03-01:
+100
Almost yet another year later. This *really* is a problem, and I figured it would have been solved by now.

This is precisely why I resorted to containing the entire Arduino environment with sophisticated shell scripts.

facchinm, 2019-03-04:
SAMD Beta package contains an updated version of openocd (0.10.0-arduino7). To test if it solves the issue, please download the core for MKRVidor, then navigate to `Arduino15/ packages/arduino/samd-beta/1.6.25/` folder and rename `boards.txt.disabled` into `boards.txt`