arduino/ArduinoCore-samd #166: Unexpected hang of SPI.begin (SERCOM::disableSPI) on SAMD21 (Adalogger M0)

PaulZC, 2016-09-24:
Hi @facchinm / @ladyada ,
This is a follow-up from issue #162 . I think it probably belongs on the Adafruit fork but that doesn't have an issue tracker...
I'm using a MAX31855 connected via SPI to an Adalogger M0. If I instantiate the MAX31855 class at the start of the sketch - before setup() - the code hangs in the call to SERCOM::disableSPI. The "while(sercom->SPI.SYNCBUSY.bit.ENABLE)" never terminates. BUT if I instantiate the MAX31855 class inside setup() it works fine... Very strange? I've condensed the problem in the attached sketch.
(I'm using Sparkfun's MAX31855 library but that isn't the cause! It's the call to SPI.begin -> SPIClass:config -> SERCOM::disableSPI that is causing grief.)
I've tried to work out a solution but this one's above my pay grade!
The work-around is to make sure SPI.begin only gets called from inside setup().
All the best, Paul

[SPI_Gremlin.zip](https://github.com/arduino/ArduinoCore-samd/files/491124/SPI_Gremlin.zip)

ladyada, 2016-09-24:
yah you cannot run SPI.begin() before setup on the SAMD, as you could with AVR
you should open an issue on the MAX31855 library and close this one. 
its not a bug with the M0, its just how it is with the more complext chipset.

PaulZC, 2016-09-24:
Thanks Ada - will do...
Please note it would be nice if you could update the Adafruit_MAX31855 library too. That has also got SPI.begin hidden in Adafruit_MAX31855::Adafruit_MAX31855(int8_t CS)!
Keep up the great work, Paul

ladyada, 2016-09-24:
yes you should open an issue on the MAX31855 library and close this one.

closed 2016-09-24 by PaulZC