arduino/ArduinoCore-samd #168: Abnormal echo on Serial

bilboquet, 2016-09-30:
Hello,
This sketch is causing me some troubles.
```
#include <Arduino.h>

void setup() {
    Serial.begin(115200);
    Serial.println("Starting !");
}

void loop() {
    int byte, count;
    while (count = Serial.available()) {
        digitalWrite(13, HIGH);
        delay(900);
        digitalWrite(13, LOW);
        delay(100);
        Serial.print("print to Serial : ");
        Serial.println(count);
        byte = Serial.read();
        Serial.print((char) byte);
    }
}
```

The board is connected to the PC (running Ubuntu 16.04) through the programming port. In a terminal I do a `cat /dev/ttyACM0` and nothing more.
For me the sketch should print: 

> Starting!

Then silently loop forever…
But it prints (blank lines have been removed):
> Starting !
> print to Serial : 3
> Sprint to Serial : 25
> tprint to Serial : 48
> aprint to Serial : 63
> rprint to Serial : 62
> tprint to Serial : 63
> iprint to Serial : 62
> nprint to Serial : 63
> gprint to Serial : 63

What is really strange is that the read characters are the printed characters… (see beginning of each line).
Somewhere there is a loop which I don't understand.
Any help will be much appreciated as this "echo" is blocking me.

Regards,

sandeepmistry, 2016-09-30:
Hi @bilboquet,

What board are you using?

bilboquet, 2016-10-05:
Hello,
Thank you very much for the reply. 
I'm on a M0 pro, flashed with the bootloader of a Zero (so I can use eclipse C++ IDE for Arduino to develop…).
Actually, I've found my mistake. I'm full of shame but if it can save someone some time I'm gonna explain it:
I've spent long hours reading the code and running it step by step with gdb… But the problem was in fact on the tty. `stty -a -F /dev/ttyACM0` showed that echo was enable on it :/ so `stty -echo /dev/ttyACM0` brought everything back to normal.

Thanks again

(PS: I still have some minor annoyances but I don't think they are related to this code)

closed 2016-10-05 by bilboquet

sandeepmistry, 2016-10-05:
@bilboquet no worries, and thanks for the follow up!

I didn't know `stty` could be using enable echo on a tty device, thanks for sharing.