arduino/ArduinoCore-samd #172: usb serial available() returning true even if buffer is empty()

tbowmo, 2016-10-09:
I discovered a fault in USB driver in SAMD 1.6.7, where a call to SerialUSB.available() always will return true, after a single character has been sent to the USB endpoint.

following test sketch will show the error:

`
void setup() {
  // put your setup code here, to run once:
  SerialUSB.begin(115200);
  while (!Serial) {}
  SerialUSB.println("Test sketch");
}

void loop() {
  // put your main code here, to run repeatedly:
  SerialUSB.print("test ");
  SerialUSB.println(millis());
  delay(1000);
  SerialUSB.println(SerialUSB.available());
  if (SerialUSB.available()) {
    SerialUSB.print(" Received ");
    SerialUSB.println(SerialUSB.read());
  }
}
`

The output with SAMD 1.6.7 package is:

`
Test sketch
test 1120
test 2120
test 3120
test 4120
test 5120
test 6120
 Received 32
test 7120
 Received -1
test 8120
 Received -1
test 9120
 Received -1
test 10120
 Received -1
test 11120
 Received -1
test 12120
 Received -1
test 13120
`

while with SAMD 1.6.6 package installed:

`
Test sketch
test 810
test 1810
test 2810
test 3810
test 4810
test 5810
 Received 32
test 6810
test 7810
test 8810
 Received 32
test 9810
test 10810
test 11810
`

The error wil result in an endless loop, if you have a while(SerialUSB.available()) {} to read characters from the serial device.

cmaglie, 2016-10-10:
@tbowmo 
thanks for the bug report, I confirm that there is a bug.

May you test if the following patch works for you?
https://github.com/arduino/ArduinoCore-samd/pull/173
https://github.com/arduino/ArduinoCore-samd/pull/173/files

tbowmo, 2016-10-10:
@cmaglie 

I have verified the patch, and it is working..

When can we expect to have this out in a release?

cmaglie, 2016-10-10:
Thanks for testing.

Expect a release very soon (maybe tomorrow).

PR #173 (2016-10-10): Fix regression: SerialUSB.available() do not return correct values
/cc @tbowmo

Fix #172
commit b06501ba (2016-10-10): Fix regression: SerialUSB.available() do not return correct values

Fix #172
closed 2016-10-10 by cmaglie