arduino/ArduinoCore-samd #175: Error compiling when using I2Cdevlib

mundodisco8, 2016-10-18:
Hi, 

I'm facing issues when trying to use I2Cdevlib in my project for the arduino Zero. Apparently, their library is expecting a constant named BUFFER_LENGTH to be present in Wire.h. This constant is present in the AVR version of the library, but not on the SAMD version. The compiler throws the following message

```
D:\Joel\Desktop\Dropbox\Arduino\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

D:\Joel\Desktop\Dropbox\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:62: error: 'BUFFER_LENGTH' was not declared in this scope

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                              ^

C:\Users\Joel\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.7\cores\arduino/Arduino.h:96:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))
```

Some code that produces this output is the following:

```
#include<Wire.h>
#include "MPU6050_6Axis_MotionApps20.h" // for the MPU6050

void setup(){
}

void loop(){
}
````

I have the latest versions of the library, the IDE (1.6.12) and the boards (1.6.8).

I opened an issue on their repo (https://github.com/jrowberg/i2cdevlib/issues/271), but I think that it might be also relevant to post one here. If you think it is not, please feel free to close it. In any case, would you be able to suggest a value for BUFFER_LENGTH? Is there a reason why not to use 32 as in the AVR version? I'm trying to use a MPU6050 accelerometer, that has a 1024 bytes buffer, in case that BUFFER_LENGTH is related to the buffer length of the target device.

facchinm, 2016-10-20:
Oh well, `BUFFER_LENGTH` is a so common name I would avoid using it in anything outside the library itself. Unfortunately AVR's Wire define is there and someone is using it. Wire for SAMD uses a 64 bytes long ringbuffer, so you could use this value if the architecture you are targeting is more powerful than AVR. I'm closing this isue since it should be solved "mainline" by i2cdev devs

closed 2016-10-20 by facchinm

mundodisco8, 2016-10-21:
Ok, Thanks.

Is just that the constant was in 1.6.6. I will patch it up modifying the library by adding it, in the meantime, so I can make sure it works as it used to be.

Cheers!

jensykesalloneword, 2016-10-25:
Hello I'm having the same issue with a Adafruit feather M0. I see from the other thread you made a change to make the buffer constant at 64. Where did you make this change? the SAMD.h file?
thanks

mundodisco8, 2016-10-26:
I answered your question in the I2Cdevlib thread, as I think it's pretty clear that is not an issue with the arduino libraries, but with the I2Cdevlib libraries.

jensykesalloneword, 2016-10-26:
Thank you. I wasn't sure if this issue thread was closed hence posting twice. thanks again.