arduino/ArduinoCore-samd #177: Uart: interrupt driven reception: How?

tuxedo0801, 2016-10-21:
Hi there,

i'm creating a sketch that is reading/writing data with I2C as well as 1wire. Beside that, I have a communication interface connected to the hardware serial.

Sending/Reading the hardware serial has a higher priority as reading/writing I2C and 1wire. I *have* to get each and every byte on UART.

The problem now is: 
Reading/writing I2C and 1wire takes a lot of time. depending on the connected device, up to several milliseconds. At 19200baud, a single bytes need ~580µs... So doing something else for several milliseconds does not sound like a good idea. But how do I overcome this? Maybe there is a receive buffer? But how much "time" can be bufferred when data is coming in with 19200 baud?
My first thought: Use an interrupt. That will "interrupt" any other "blocking" thing in loop(), let me receive the data, and continue with loop().

Arduino for 328p and 32u4 is looking in the main() for new data and calls SerialEvent() before/after loop() is called. So there is no real "interrupt driven data reception". If loop blocks for 10ms, I may miss a lot of bytes on UART.

I guess this is the same for samd21?

So my question is: Is there, for SAMD21, a way to get notified by a real interrupt when data is received on UART? If yes: Could you please point me to some helpful - beginner friendly - documentation?

dlabun, 2016-10-21:
This is the wrong place to ask for support, you need to post questions like this to the Arudino forums

tuxedo0801, 2016-10-24:
Issue solved:

http://forum.arduino.cc/index.php?board=98.0

Solution:
* use non-default sercom and add own routine to SERCOMx_Handler() function

or

* use a time that is slightly faster than receiving time for 1 byte.

closed 2016-10-24 by tuxedo0801