arduino/ArduinoCore-samd #181: analogWriteResolution without function?

Adminius, 2016-11-09:
Hello,
i'm disappointed about analogWriteResolution function...
as you can see [https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/wiring_analog.c#L215](url)
the input value is mapped to 8 bit value, so it doesn't matter wich resolution is set by analogWriteResolution()

can you check this, pls?
Thx

tuxedo0801, 2017-03-13:
@cmaglie & Co... 
No comments so far?

cmaglie, 2017-03-13:
The purpose of the `analogWriteResolution(..)` is to provide a portable API so you can write a sketch like:

```
analogWriteResolution(10);
analogWrite(1, 600);
```

and it will work consistently across all platform/boards (even different CPU architectures!) and at the same time deliver the maximum accuracy available on each board.
Of course if you exceed the hardware capabilities the value passed in analogWrite will be truncated, writing `analogWriteResolution(100)` will not magically give you a PWM with 100 bits of resolution...
The SAMD21 has 8-bit PWMs, so the values are mapped to 8 bit to match hardware capabilities.

Adminius, 2017-03-14:
All PWM TCC und TC timers (0-5) are 16bit timer and not 8bit. But analogWrite can only 8bit...
It would be nice, if at least TCC timer could be tuned to at least 12bit resolution. (TC timer can only 8 or 16 :( )

cmaglie, 2017-03-15:
I see, I've changed the issue subject accordingly.

@Adminius can you provide a PR for that?

Adminius, 2017-04-05:
@cmaglie with this PR it ist possible to set resolution with analogWriteResolution up to 16 bit and set analogWrite values up to 65535.
PWM frequency is still 732 Hz

Adminius, 2017-06-21:
@cmaglie, @facchinm any news?

cmaglie, 2017-07-11:
Merged #226, thank you!

closed 2017-07-11 by cmaglie