arduino/ArduinoCore-samd #187: OpenOCD debugging session erases bootloader

facchinm, 2016-12-14:
A typical debugging session with openocd + gdb looks like

```
arm-none-eabi-gdb  filename.ino.elf eval-command "target remote |  \"openocd" -c \"gdb_port pipe\" --search \"...openocd/scripts/\" --file \".../openocd_scripts/arduino_zero.cfg\""
load
monitor reset halt
monitor reset init
continue
```

Unfortunately, this ends up with the bootloader being wiped out and the registers containing dummy data.

If we relocate the sketch to flash position 0x4000 instead than using 0x2000, everything works perfectly.

From [http://forum.arduino.cc/index.php?topic=330338.msg2590441#msg2590441](http://forum.arduino.cc/index.php?topic=330338.msg2590441#msg2590441) it appears that .org boards are immune to this problem.
Looking at the M0 pro linker script, the sketch starts at 0x4000 and the board is debuggable.

The flash organization is 

```
#0 : at91samd at 0x00000000, size 0x00040000, buswidth 1, chipwidth 1
    #  0: 0x00000000 (0x4000 16kB) not protected
    #  1: 0x00004000 (0x4000 16kB) not protected
    #  2: 0x00008000 (0x4000 16kB) not protected
    #  3: 0x0000c000 (0x4000 16kB) not protected
    #  4: 0x00010000 (0x4000 16kB) not protected
    #  5: 0x00014000 (0x4000 16kB) not protected
```

Should we patch openocd maybe? Or patching the bootloader relocation could be a better long time strategy (ld scripts are in the core, so I'd avoid this second option)

arduino/OpenOCD#2 could be a solution, but it is untested on smaller SAMD... Should we open a PR to mainline openocd and see what people say?

dlabun, 2016-12-14:
Personally I think this should be treated as an OpenOCD issue as I can debug without issues within Atmel Studio.

chrise96, 2017-03-02:
Same issue here with the Arduino Zero .cc version. Board is dead when trying to debug. I can debug my board using the .org Arduino M0 pro bootloader (Bootloader_d21 - stk500v2)...

facchinm, 2017-03-02:
Solved by merging arduino/OpenOCD#2 , which will become available in next samd release

chrise96, 2017-03-02:
uhhh get the error "src/target/aarch64.c:33:9: error: '__unused' macro redefined" when make openocd. When will this be released?

facchinm, 2017-03-03:
This is the build script repo https://github.com/arduino/OpenOCD-build-script ; try it before doing anything from the scratch :smile: 
The updated tool will be officially released during this month.

chrise96, 2017-03-17:
Still no luck with debugging the Arduino zero. So, I made a video to clarify the issue.

https://youtu.be/VsR1WFR9vbw 
1.	Bootloader from the Arduino M0 pro is working (0:04)
2.	Arduino Zero (sam-ba) debugging session erases bootloader (0:20)
3.	Using the latest OpenOCD github build (1:27)

facchinm, 2017-03-17:
Uh, it depends on the Openocd which is being used by platformIO IDE (I can't spot it from the video). Please [download](http://downloads.arduino.cc/tools/openocd-0.9.0-arduino6-static-x86_64-apple-darwin15.6.0.tar.bz2) the latest binary for osx, please follow the following steps and report the results:
- flash the .cc bootloader
- open a shell
- execute 
```
$SAMD_CORE_TOOLS_PATH/arm-none-eabi-gdb  $FILENAME.ino.elf eval-command "target remote |  \"$EXTRACTED_OPENOCD_PATH/openocd" -c \"gdb_port pipe\" --search \"$EXTRACTED_OPENOCD_PATH/openocd/scripts/\" --file \"$EXTRACTED_OPENOCD_PATH/openocd_scripts/arduino_zero.cfg\""
load
monitor reset halt
monitor reset init
continue
```
obviously changing the $ stuff with the actual paths. If you load a blink, the led should start blinking and you can continue the debug session (inserting breakpoints etc)

chrise96, 2017-03-18:
The latest OpenOCD works with Arduino Zero using the command line and also with Eclipse! Nice work, thanks for your help 😃 👍

facchinm, 2017-07-27:
Closing as fixed

closed 2017-07-27 by facchinm