arduino/ArduinoCore-samd #199: Changing trigger level in attach interrupt during runtime

joverbee, 2017-01-07:
Current behaviour in Winterrupts.c assumes we perform attachinterrupt(pin,ISR,mode) only once after reset. However if you want to dynamically change the trigger mode, this leads to wrong behaviour:
e.g first starting with RISING and then FALLING leads actually to BOTH because the mode bits are ORed into the register. 

I assume putting this in front of the switch(mode) in attachinterrupt solves the issue
....
EIC->CONFIG[config].reg &=~ (EIC_CONFIG_SENSE0_Msk << pos); //reset sense mode
  switch (mode) 
....
However I totally fail to explain my setup that it should recompile this file and use my own version rather than the default one? Any help would be appreciated...its driving me crazy (I seem to always get stuck for hours on include issues...aaargh)

facchinm, 2017-01-09:
Hi @joverbee ,
if you modified the right file, it should be automatically recompiled. What do you mean with "I seem to always get stuck for hours on include issues" ? However, the patch looks good, comments on its own thread

joverbee, 2017-01-09:
I figured it out. I am working on Mac and it wasn't really clear where the actual Winterrupts.c file was that the ide was using (I had a few different one scattered over my drive). I found out by trying to redefine EIC_Handler in the arduino code which then made the compiler complain that he already had a definition in this-that-directory.

cmaglie, 2017-09-04:
Thanks! Fixed by a9fa694.

closed 2017-09-04 by cmaglie