arduino/ArduinoCore-samd #200: External interrupts take 6 us to responds

joverbee, 2017-01-08:
Even though we can expect the interrupt handling through arduino interface to be slower than doing low level coding with the dataheet at hand, 6 us is really ridicoulously long for a processor with 20ns clock cycle (300 clock cycles).
I havent figured out yet to find the cause of this (played with genclck of the EIC but no improvement) but to me it is clear that this can not be an optimal way to use the EIC controller. This makes interrupting for anything dynamic impossible. It seems even longer than it was on arduino uno boards, but dont recall the exact time.
( btw I managed through hw event system to start stop the TCC on external interrupt in 180ns=9 cycles- but of course we need to do more checking and jumping to ISR in this case, but still: does it really take 291 clock cycles to do this?)

joverbee, 2017-01-08:
i checked the code and the culprit is the EIC handler which checks all 16 interrupt channels even when they are not enabled. As I was working with extint14 this means my ISR is preced by 14 reads of the status register which is a total waste of time.
I plan to rewrite the code in Winterrupts.c to make a stack of attached interupts with their masks for checking the status bit. i would then service the isrs on a first come first serve basis. This should dramatically speed up interrupts especially when not all 16 channels are in use ( that would be a very rare case anyway)

Can anyone point me to a simple description on how I have to tell the IDE to work with my version of the core rather than the one supplied ? THis would really allow me to focus on this issue rather than on figuring out how to make it compile.

dlabun, 2017-01-12:
The quick and dirt method is to just modify the copy of the core installed on your computer... The only draw back is if you update the core through the Board Manager you wipe out all over your modifications.

joverbee, 2017-01-12:
That is what I figured out doing. But I also made a pull request, so hopefully the next version that comes in through the board manager will contain my changes and we all profit ;)

cmaglie, 2017-09-14:
Fixed by #261

closed 2017-09-14 by cmaglie