arduino/ArduinoCore-samd #204: Bootloader Size

Sorunome, 2017-01-16:
Hey, I tried making a custom bootloader for the arduino zero with a different size, though.

I updated the size in https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/bootloader_samd21x18.ld#L29 and https://github.com/arduino/ArduinoCore-samd/blob/master/platform.txt#L155 and https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/linker_scripts/gcc/flash_with_bootloader.ld#L28

When uploading via Programming Port everything seems to be working fine, however when uploading via Native Port BOSSA still issues chipErase(addr=0x2000) and writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)

Here's the full output, in hope somebody could help me:

```
Set binary mode
readWord(addr=0)=0x20007ffc
SAM-BA operation failed
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v2.0 [Arduino:XYZ] Jan 16 2017 20:39:51
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Chip ID      : 10010005
version()=v2.0 [Arduino:XYZ] Jan 16 2017 20:39:51
Version      : v2.0 [Arduino:XYZ] Jan 16 2017 20:39:51
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
An error occurred while uploading the sketch
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.002 seconds
 
Write 9752 bytes to flash (153 pages)
write(addr=0x20005000,size=0x1000)
writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)
```

Any help would be greatly appreciated!

facchinm, 2017-01-17:
Hi @Sorunome , the issue is already in the todo list on `bossac` repo https://github.com/shumatech/BOSSA/issues/33 . Feel free to comment there to speedup the process :slightly_smiling_face: Thanks!

closed 2017-01-17 by facchinm

frippe75, 2018-06-20:
Opened https://github.com/arduino/Arduino/issues/7709