arduino/ArduinoCore-samd #206: Invalid driver signature for CircuitPlay Express

dlabun, 2017-01-22:
The driver packaged within the SAMD core for the Adafruit CircuitPlay Express does not have a valid signature, causing Windows validation errors when you attempt to install 1.6.10 or 1.6.11 of the core. There were a bunch of drivers on the Arduino repository that were fixed, looks like this driver needs updating as well.

facchinm, 2017-01-23:
Thanks for spotting this! Backporting https://github.com/arduino/Arduino/commit/123c231c8e8c9b1d610d4ad26a294830c624e1bc should fix the issue @cmaglie

cmaglie, 2017-01-23:
> Backporting arduino/Arduino@123c231 should fix the issue

no, the drivers are different, it requires to be re-sign from Adafruit

/cc @ladyada

dlabun, 2017-01-23:
On a related side note, What board is this driver even intended for? I couldn't find anything on the Adafruit website, making me wonder if this driver is even necessary.

ladyada, 2017-01-23:
[cplayexpress.zip](https://github.com/arduino/ArduinoCore-samd/files/724915/cplayexpress.zip)

here's latest signed driver. for Unknown Reasons i couldnt rename cdc_vista.cat without it complaining about the signature :/

cmaglie, can you verify for me that you get no warnings? thx :)

cmaglie, 2017-01-27:
it works, thanks!

commit cb9927fd (2017-01-27): Fixed Circuit Playground Express driver

Fix #206
closed 2017-01-27 by cmaglie