arduino/ArduinoCore-samd #209: Warning on wiring.c

jwillemsen, 2017-02-02:
We get a warning using platformio with the sodaq boards. Looks the warning is coming from this repo

I am trying to enable travis-ci support using pipeline for the sodaq boards. We want our code to be warning free so use -Werror as option, but it seems there is a warning in the sodaq one support code.

```
Project has been successfully initialized!

Useful commands:

`platformio run` - process/build project from the current directory

`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board

`platformio run --target clean` - clean project (remove compiled files)

`platformio run --help` - additional information

[Thu Feb  2 20:03:34 2017] Processing sodaq_autonomo (platform: atmelsam, board: sodaq_autonomo, framework: arduino)

--------------------------------------------------------------------------------

PackageManager: Installing framework-arduinosam @ ~2.10611.0

Downloading...

Unpacking...

framework-arduinosam @ 2.10611.0 has been successfully installed!

Verbose mode can be enabled via `-v, --verbose` option

Converting SodaqAutonomo-basic.ino

Collected 6 compatible libraries

Looking for dependencies...

Library Dependency Graph

|-- <src>

Compiling .pioenvs/sodaq_autonomo/src/SodaqAutonomo-basic.ino.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduinoVariant/variant.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/IPAddress.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/Print.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/Reset.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/RingBuffer.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/SERCOM.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/Stream.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/Tone.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/USB/CDC.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/USB/PluggableUSB.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/USB/USBCore.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/USB/samd21_host.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/Uart.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/WInterrupts.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/WMath.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/WString.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/abi.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/avr/dtostrf.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/cortex_handlers.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/delay.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/hooks.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/itoa.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/main.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/new.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/pulse.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/pulse_asm.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/startup.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/wiring.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/wiring_analog.o

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/wiring_digital.o

/home/travis/.platformio/packages/framework-arduinosam/cores/sodaq/wiring.c: In function 'init':

/home/travis/.platformio/packages/framework-arduinosam/cores/sodaq/wiring.c:60:12: error: unused variable 'ul' [-Werror=unused-variable]

uint32_t ul ;

^

cc1: all warnings being treated as errors

Compiling .pioenvs/sodaq_autonomo/FrameworkArduino/wiring_private.o

*** [.pioenvs/sodaq_autonomo/FrameworkArduino/wiring.o] Error 1
```

facchinm, 2017-02-07:
Hi @jwillemsen ,
it appears that only the sodaq core ( https://github.com/SodaqMoja/SodaqCore-samd ) is affected; it also appears they don't specify `-Werror=unused-variable` and `ul` variable is indeed used, so I believe the bug is somewhere in plaformio. Please report the bug there, thanks!

closed 2017-02-07 by facchinm

jwillemsen, 2017-02-07:
Thanks for the feedback @facchinm, created an issue for platformio, not sure which code they exactly compile, I assumed from this repo but that sounds a wrong assumption