arduino/ArduinoCore-samd #213: Should WString use memcpy not stycpy

petejan, 2017-02-25:
In WString.cpp (this also applies to avr core)

https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/WString.cpp#L179 and
https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/WString.cpp#L271

strcpy is used, where are length is passed to the copy and concat operation, whereas memcpy should be used so the char* can contain the 0 character.

Reneg973, 2021-04-07:
Well AFAIK we talk about null terminated strings? So why should a null in this case not terminate it?
Of course there are sometimes strings, which are split by a "\0" and finalized by a "\0\0". In this case String is not the correct object to handle it.

matthijskooijman, 2021-04-07:
Usually, strings will be NUL-terminated, but the String class already stores an explicit length and supports passing a length to some methods and constructors, which means it *should* be able to support strings with embedded NUL characters too. Note that the String class is now developed as part of the ArdunoCore-API repo, so this issue should probably be discussed there. Some changes wrt to supporting strings with NUL have been made already in https://github.com/arduino/ArduinoCore-API/pull/97, and some remaining issues are tracked in https://github.com/arduino/ArduinoCore-API/issue/111 https://github.com/arduino/ArduinoCore-API/issue/112 (though those are probably not all issues, maybe a container "support strings with embedded NUL" issue would make sense).

Maybe this issue could be closed in favor of the existing (or a new) issues in the -API repo?