arduino/ArduinoCore-samd #214: OSC8M now runs at 4MHz

keestux, 2017-02-27:
Since we switched to using CMSIS 4.5 we are not initializing OSC8M at 8MHz anymore. It is now at 4MHz.

Take a look at this. Although the startup.c code did not change, unfortunately the defines for the `SYSCTRL->OSC8M.bit.PRESC` bitfield did. The code in startup.c is like this:

    /* ----------------------------------------------------------------------------------------------
     * 6) Modify PRESCaler value of OSC8M to have 8MHz
     */
    SYSCTRL->OSC8M.bit.PRESC = SYSCTRL_OSC8M_PRESC_1_Val ;
    SYSCTRL->OSC8M.bit.ONDEMAND = 0 ;

That define `SYSCTRL_OSC8M_PRESC_1_Val` is the trouble maker. Before it was:

    #define   SYSCTRL_OSC8M_PRESC_1_Val       0x0u   /**< \brief (SYSCTRL_OSC8M) 1 */
    #define   SYSCTRL_OSC8M_PRESC_2_Val       0x1u   /**< \brief (SYSCTRL_OSC8M) 2 */
    #define   SYSCTRL_OSC8M_PRESC_4_Val       0x2u   /**< \brief (SYSCTRL_OSC8M) 4 */
    #define   SYSCTRL_OSC8M_PRESC_8_Val       0x3u   /**< \brief (SYSCTRL_OSC8M) 8 */

but now it is:

    #define   SYSCTRL_OSC8M_PRESC_0_Val       0x0ul  /**< \brief (SYSCTRL_OSC8M) 1 */
    #define   SYSCTRL_OSC8M_PRESC_1_Val       0x1ul  /**< \brief (SYSCTRL_OSC8M) 2 */
    #define   SYSCTRL_OSC8M_PRESC_2_Val       0x2ul  /**< \brief (SYSCTRL_OSC8M) 4 */
    #define   SYSCTRL_OSC8M_PRESC_3_Val       0x3ul  /**< \brief (SYSCTRL_OSC8M) 8 */

That's a nasty change from Atmel, isn't it?

facchinm, 2017-02-28:
Hi @keestux ,
indeed it's a very nasty change... Fixed by 4ac0cc1df0f0018791437a4bb498c7ec6bf60edf, thanks for reporting

closed 2017-02-28 by facchinm

keestux, 2017-02-28:
Do you happen to know the relation between CMSIS 4.x versions and ASF versions?
I'm unhappy with Atmel's update. If they can screw up like this, there could be more.

cmaglie, 2017-03-01:
The CMSIS package is made of various parts:

```
CMSIS/CMSIS/...
CMSIS/Device/ARM/...
CMSIS/Device/Atmel/...
```

The interesting part is the `Devices/Atmel` folder that is like an add-on maintained exclusively by Atmel-Microchip.

With the last update a bunch of things changed without backward compatibility, for example in those commits:

https://github.com/arduino/ArduinoCore-samd/commit/7ebd62a841e2a42837aad62f955f4300fe0b9eaf
https://github.com/arduino/ArduinoCore-samd/commit/68e30401db4eaa2b250006af44d1707961540d13
https://github.com/arduino/ArduinoCore-samd/commit/2f3b976652050c6a280a1854e0a7429000eb826b
https://github.com/arduino/ArduinoCore-samd/commit/80f629c379fb15c7e7e4d1ae2becdbbb3c81f6cc

In all the cases above we were lucky since the change in CMSIS produced a compiler error so it was easy to find and fix. Even if annoying I can still tolerate this kind of backward incompatible changes.

In the case reported in this PR, instead, the name of the constant didn't change but changed his meaning, this is a far more dangerous change since it doesn't produce compiler error or warning but introduce a runtime error that may be very difficult to diagnose.

keestux, 2017-03-01:
What I wanted to know is: what CMSIS source did you use? Do you have an URL? In fact, I'm curious as to know how the following packages are being assembled/created. Is there a GIT repo for them? (Name and version how they appear in package_index.json)

- CMSIS (versions: 4.0.0-atmel, 4.5.0)
- CMSIS-Atmel (versions: 1.0.0, 1.1.0)

All I could find myself is ASF (Atmel Software Framework), see http://asf.atmel.com or http://www.atmel.com/asf  which is now at version v3.33. It contains CMSIS too.

sandeepmistry, 2017-03-01:
> What I wanted to know is: what CMSIS source did you use? Do you have an URL? 

@aethaniel can you help answer this?

@keestux Git repos are here:
 * https://github.com/arduino/ArduinoModule-CMSIS
 * https://github.com/arduino/ArduinoModule-CMSIS-Atmel

keestux, 2017-03-01:
> In the case reported in this PR, instead, the name of the constant didn't change but changed his meaning, this is a far more dangerous change since it doesn't produce compiler error or warning but introduce a runtime error that may be very difficult to diagnose.

Yeah and if you look this particular change there is something I want to get of my chest. (Not blaming you or the Arduino community, just know know who.)

1. to change the value of a define you need very good reasons (bugs)
2. it is horrifying to see what the new names and the new values are

The old names where saying something about the divide factor. The new names describe their value instead of their meaning. Now there is no added value to use the defines anymore. In fact, they made you wish we had used hard coded numbers.

    SYSCTRL->OSC8M.bit.PRESC = 0;   // divide by 1

keestux, 2017-03-01:
@sandeepmistry Thanks for the pointers
That raises new questions though. Hopefully @aethaniel knows it all.

1. CMSIS-Atmel, the repo is populated with a commit message "Initial import", but it does not say where the import came from. Is it ASF, and if so, which version? Maybe Atmel Studio, but that just contains ASF, I think.

2. ArduinoModule-CMSIS does not contain version 4.0.0-atmel. Where did version 4.0.0-atmel come from? This package contains all the samd include files (and others). Although this is history, it is still interesting to know if there is a repo. Was that imported from ASF, and if so, what version?

aethaniel, 2017-03-01:
>What I wanted to know is: what CMSIS source did you use? Do you have an URL?

CMSIS is provided by ARM on Github.
The header package containing Atmel product definitions has been more or less extracted from Atmel DFPs at different times for Arduino dedicated use, as far as I remember. Cf the repos URL listed above by Sandeep.

It is indeed very unfortunate to meet that kind of issue. I don't what else to say except each update must be done carefully with double check for any doubt. In that case, the product datasheet is the most reliable source.

cmaglie, 2017-03-01:
Just to sort things out of the packages contents available in the package_index.json:

`CMSIS` version `4.0.0-atmel` - is the CMSIS distributed by Atmel some years ago through Atmel Studio. This package contains both CMSIS 4.0.0 **and** the Atmel extensions, that's why it's version is **4.0.0-atmel** and not just 4.0.0. This is the oldest package and at the times it wasn't tracked in git.

`CMSIS` version `4.5.0` - is the plain CMSIS 4.5.0 **without** Atmel extensions. This is tracked in https://github.com/arduino/ArduinoModule-CMSIS.

`CMSIS-Atmel` version `1.0.0` and `1.1.0` - is the Atmel's CMSIS extension (basically just the content of `CMSIS/Device/Atmel/...`) . Since Atmel doesn't provide his own versioning we just started from 1.0.0, and updated to 1.1.0 after adding definitions for SAML21. This is tracked in https://github.com/arduino/ArduinoModule-CMSIS-Atmel

We've split the Atmel extensions from CMSIS to better handle updates and possibly allow other vendors to provide their own extensions.

keestux, 2017-03-01:
Thanks @aethaniel and @cmaglie 

I suggest that, for CMSIS-Atmel, we now only look at ASF (if there are any updates). And document which ASF version we use for the next update. The only remaining thing would be to know which ASF version was used for 1.0.0 and 1.1.0

And, of course, look very carefully before actually taking any update :-)

keestux, 2017-03-02:
Just for completeness of my investigation I want to mention these changes (which we don't use)

    #define   WDT_CONFIG_PER_8_Val            0x8
    #define WDT_CONFIG_PER_8            (WDT_CONFIG_PER_8_Val          << WDT_CONFIG_PER_Pos)

versus

    #define   WDT_CONFIG_PER_8_Val            0x0
    #define WDT_CONFIG_PER_8            (WDT_CONFIG_PER_8_Val          << WDT_CONFIG_PER_Pos)

Same for `WDT_CONFIG_WINDOW_8_Val` and `WDT_EWCTRL_EWOFFSET_8_Val`.

`TCC0_TYPE` and `TCC1_TYPE` changed too, but we don't use these.