arduino/ArduinoCore-samd #215: recover MKR1000 board automatically

hanabanana, 2017-02-28:
It is observed that programming the MKR1000 board via IDE easily fails at upload and leaves the board in bootloader mode. When it happens, a manual reset involving pressing the reset button is almost always needed to recover the board so that it can be programmed again. However, this approach does not allow us to run automated tests. Any idea about making it occur less often? Also, is there an automatic way to reset the board when this occurs? 

Thanks.

facchinm, 2017-03-01:
Hi @hanabanana ,
usually the situation you are describing is due to bossac trying to access the CDC device too early. AFAIR @sandeepmistry provided a patch for the bootloader to ensure this "transition" process is faster and the CDC shows up when everything is setup. It should be included in the 2.0 revision of the bootloader (the one with the fading led during bootloader mode), any chance you could test again in your setup after updating? Is it feasible?

sandeepmistry, 2017-03-01:
@facchinm right, this was in https://github.com/arduino/ArduinoCore-samd/commit/7a23f8d484170fff315639e80854f16264a7d9e6#diff-ee8867e3b41c9a75b09a20c63734d201.

It's a bit harder to flash the MKR1000 boot loader compared to the Zero. @facchinm do you think we should make a sketch to do this?

facchinm, 2017-03-01:
Well, unless we protect the boot sectors (which we don't do ATM), it's super easy to rewrite a bootloader live (although it's everything but safe :smile: ). If we add a sketch we must write in big capital letters that it's dangerous if you don't own an external programmer

hanabanana, 2017-03-01:
@facchinm Not sure when the patch was released(is it from 1.6.9 samd?).  I checked the file in 1.8.0 IDE+1.6.11 samd and I can see those changes. So I am assuming the corresponding compiled bootloader  is the 2.0 version or later. Please correct me if wrong.  
The boards were bought fairly recently but they do not seem to have the fading LED in bootloader mode. So I am going to reflash the bootloader with the one in 1.6.11 samd and see if it gets better.

facchinm, 2017-03-01:
Correct in every point :slightly_smiling_face: The MKR1000 we are shipping right now were flashed before the revamped bootloader was declared stable, so they contain the old one. The MKRZero instead already comes with the new bootloader.

hanabanana, 2017-03-06:
With the updated bootloader, the boards still got stuck eventually after programming it several times. We run the same tests on Uno and Due and they have been very robust. It seems that the board does not handle consecutive programming well in particular. For now, I had to disable those tests on this board.

facchinm, 2017-03-07:
Well, both UNO and DUE have a separate microcontroller to handle the upload, it never disconnects from the host PC so it's much less dependent on host connection/disconnection routines. Could you share a sample of the upload stress test you are running? Thanks!

hanabanana, 2017-03-07:
Each test compiles a server that includes a predefined set of libraries and then uploads the server to board. This is done via either the Arduino CLI or a Makefile+bossac and they run once per day. Right now, I've taken out the stress test and limited to only four feature tests with each uploads once at start and runs the remaining test(read/write to a sensor for example) for a couple minutes before the next test runs. Yet, the board still gets stuck. One of my colleagues mentions that, with a Genuino MKR1000 he was able to open the port at 1200 and restored it back but with Arduino MKR1000 nothing helped other than pressing the reset. I am not sure if this is entirely true. Overall, this issue does not occur everytime but it is to a point that is often enough to prevent us from doing any automated testing. I also want to mention that the issue occurs even more frequently on Linux on which we have not been able to run any tests.

Here's one log when it fails to program,
Sketch uses 56,096 bytes (21%) of program storage space. Maximum is 262,144 bytes.
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
PORTS {COM1, COM2, COM4, } / {COM1, COM2, COM4, } => {}
Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.

hanabanana, 2017-03-09:
One thing I'd like to point out is before each test, we restart the computer. Though the driver has already been installed, I wonder if this could cause unusual delay in connection/disconnection for first use that makes it unstable? Just a thought.

facchinm, 2017-03-09:
It shouldn't, unless at boot the PC tries to configure the port in strange ways (eg. opening at 1200bps).
However, from your log, it appears that the bootloader port is showing but it's not getting the reset. This is a known issue, which the IDE overcomes by selecting the user selected port if none of the system ports disappears/reappears. Could it be the case? The relevant code dealing with this problem is https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L166

hanabanana, 2017-03-15:
I did see IDE trying to use the user selected port for upload when that happens but probably not the CLI. This is IDE 1.6.13. Could that have avoided the getting stuck in bootloader? Even if the user selected port is used, my understanding is that the upload won't work.