arduino/ArduinoCore-samd #218: USB "Unrecognized device" ATSAMW25 custom board

choncba, 2017-03-08:
I'm working on a prototype with ATSAMW25, the same one used on the MKR1000.
I use the SWD port connected to a Raspberry Pi with OpenOCD to load the bootloader (samd21_sam_ba_arduino_mkrzero.bin), and the process completes without errors.
The problem is that when connecting to the PC via USB, Windows does not recognize the device.
The hardware is identical to the MKR1000, except that it does not have the battery support stage.
Is it necessary to perform any additional configuration in the module, or in the bootloader already compiled?

facchinm, 2017-03-08:
Hi @choncba ,
the W25 is essentially a SAMD21 with an onboard WINC1500 chip, so any bootloader for SAMD21 boards should work. All our software (bootloader + core) is configured to use an external 32KHz quartz, did you include it or are you clocking with the internal RC? In the latter case you'd need to slightly modify the clock sequence to allow booting without the quartz (check the `CRYSTALLESS` define in `startup.c` for reference).

Instead, if you have the 32K crystal mounted, make sure you are running a recent version of openocd. Older ones are buggy and could mess up the fuse values, leading to a never-ending watchdog reboot. https://github.com/arduino/OpenOCD/commit/76b0521e59cb535bbdf36ccf328412ccdd933a20 and https://github.com/arduino/OpenOCD/commit/688b0b6cc2968fff1823cd47d245354af28dcaef should solve this problem. You can download the precompiled openocd with the patches applied for rPi from http://downloads.arduino.cc/tools/openocd-0.9.0-arduino5-static-arm-linux-gnueabihf.tar.bz2

choncba, 2017-03-08:
Hi @facchinm, I really appreciate your answer.
I compiled OpenOCD directly from the source in the Raspberry Pi with:
 **git clone git: //git.code.sf.net/p/openocd/code openocd-code**
Following this tutorial: [https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/wiring-and-test](url)
Is the precompiled version you attached different than this one?
Or maybe the OpenOCD configuration used in that tutorial is different?

The configuration file I use in OpenOCD (openocd.cfg) is:

```
Source [find interface / raspberrypi2-native.cfg]
Transport select swd
 
Set CHIPNAME at91samd21g18
Source [find target / at91samdXX.cfg]
 
# Did not yet manage to make a working setup using srst
#reset_config srst_only
#reset_config srst_nogate
 
Adapter_nsrst_delay 100
Adapter_nsrst_assert_width 100
 
Init
Targets
Reset halt
At91samd bootloader 0
Program samd21_sam_ba_genuino_mkr1000.bin verify
At91samd bootloader 8192
Reset
Shutdown
```
It was necessary to comment **#reset_config srst_nogate** because it generates error using the reset pin in my Rpi2

Regards.-

choncba, 2017-03-09:
In case it works, I attach the result of OpenOCD when programming the bootloader:
```
Open On-Chip Debugger 0.10.0+dev-00093-g6b2acc0 (2017-03-06-12:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* at91samd21g18.cpu  cortex_m   little at91samd21g18.cpu  running
Info : SWD DPIDR 0x0bc11477
target halted due to debug-request, current mode: Thread
xPSR: 0x41000000 pc: 0x000001b0 msp: 0x20007fcc
Info : SWD DPIDR 0x0bc11477
target halted due to debug-request, current mode: Thread
xPSR: 0x41000000 pc: 0x0000020c msp: 0x20007fcc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 16384 bytes from file samd21_sam_ba_arduino_mkr1000.bin in 0.658182s (24.309 KiB/s)
** Programming Finished **
** Verify Started **
verified 6404 bytes in 0.027391s (228.320 KiB/s)
** Verified OK **
Info : SWD DPIDR 0x0bc11477
shutdown command invoked
```

facchinm, 2017-03-09:
The problem is due to `At91samd bootloader 8192` line (bootloader locking).
Please download the avrdude for rPi I linked and issue a `at91samd restore` command using it. This should restore the fuses to safe values and allow you to recognize the module.

choncba, 2017-03-09:
Could you please tell me how to install the openocd version you attached?
I unzipped the files in /home/pi, but there is no makefile, I have already uninstalled the version 0.10 I had originally.
After doing so, should I edit the .cfg file with the command that you indicate?

facchinm, 2017-03-10:
Version is 0.9.0 with a couple of patches applied. It's already compiled so you don't need to compile it yourself. Once extracted, you can launch the restore with a cfg like this:
```
Source [find interface / raspberrypi2-native.cfg]
Transport select swd

Set CHIPNAME at91samd21g18
Source [find target / at91samdXX.cfg]

Adapter_nsrst_delay 100
Adapter_nsrst_assert_width 100
 
Init
Targets
Reset halt
at91samd restore
Reset
```

using the same commandline of the openocd you already installed (just replace `openocd` with `place_you_extracted/bin/openocd` )

choncba, 2017-03-10:
I get this error now:
```
root@raspberrypi:/home/pi/bootloader# openocd
Open On-Chip Debugger 0.10.0+dev-00093-g6b2acc0 (2017-03-10-15:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
trst_and_srst separate srst_gates_jtag trst_push_pull srst_push_pull connect_deassert_srst
trst_and_srst srst_pulls_trst srst_gates_jtag trst_push_pull srst_push_pull connect_deassert_srst
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
trst_and_srst srst_pulls_trst srst_nogate trst_push_pull srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 400 kHz
Info : SWD DPIDR 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* at91samd21g18.cpu  cortex_m   little at91samd21g18.cpu  running
Error: can't request a halt while in reset if nSRST pulls nTRST
in procedure 'reset' called at file "openocd.cfg", line 39
in procedure 'ocd_bouncer'

Error: Failed to enable the FPB
Polling target at91samd21g18.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 100ms
Error: at91samd21g18.cpu -- clearing lockup after double fault
Polling target at91samd21g18.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 300ms
Error: at91samd21g18.cpu -- clearing lockup after double fault
Polling target at91samd21g18.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 700ms
Error: at91samd21g18.cpu -- clearing lockup after double fault
Polling target at91samd21g18.cpu failed, trying to reexamine
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 1500ms
```

And keeps polling...

facchinm, 2017-03-13:
You are not running the openocd packaged by us (which is 0.9.0). Repeat the procedure with the right binary please

choncba, 2017-03-14:
Apologies, my mistake The problem was that my rpi was confusing the official openocd installation with the modified one. 
But, anyway, something is missing.
If I execute this command : `root@raspberrypi:/home/pi/bootloader# /home/pi/openocd-arduino/bin/openocd `

where, /bootloader contains the **openocd.cfg** file:
```
source [find /home/pi/openocd-arduino/share/openocd/scripts/interface/raspberrypi-native.cfg]
transport select swd

set CHIPNAME at91samd21g18
source [find /home/pi/openocd-arduino/share/openocd/scripts/target/at91samdXX.cfg]

adapter_nsrst_delay 100
adapter_nsrst_assert_width 100
 
init
targets
reset halt
at91samd restore
reset
```

I get:

```
root@raspberrypi:/home/pi/bootloader# /home/pi/openocd-arduino/bin/openocd
Open On-Chip Debugger 0.9.0-g688b0b6 (2017-01-12-15:07)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (bcm2835gpio)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: usbprog
4: jlink
5: vsllink
6: rlink
7: ulink
8: arm-jtag-ew
9: hla
10: osbdm
11: opendous
12: aice
13: cmsis-dap

Error: Debug Adapter has to be specified, see "interface" command
```
So apparently this patched version was not compiled for raspberry pi (?), And I could not figure out how to set up a new adapter to this. Any idea?

**Note: If I don't specify the folder address of the target and interface in openocd.cfg file, I get an "can`t find file error"**

facchinm, 2017-03-14:
The packet we provide doesn't include `raspberrypi-native.cfg` file, so you must specify the old installation path in the cfg file in place of
```
source [find /home/pi/openocd-arduino/share/openocd/scripts/interface/raspberrypi-native.cfg]
```

choncba, 2017-03-14:
Thanks for the patience, I get the same result with 
`source [find /usr/local/share/openocd/scripts/interface/raspberrypi2-native.cfg]`
Same with `raspberrypi-native.cfg` and `sysfsgpio-raspberrypi.cfg`
It looks as if it has not been configured with `./configure --enable-sysfsgpio --enable -bcm2835gpio` when compiled.

choncba, 2017-03-20:
Could you tell me what are the arrangements that were made to copy them in the new version before compiling it? The version adapted here is not compatible with the Raspberry Pi 2.

Regards.

facchinm, 2017-03-20:
There are surely some more patch to add, but the starting repo is https://github.com/arduino/OpenOCD-build-script/tree/static . In theory, you simply need to add these two patches (https://github.com/arduino/OpenOCD/commit/76b0521e59cb535bbdf36ccf328412ccdd933a20 and https://github.com/arduino/OpenOCD/commit/688b0b6cc2968fff1823cd47d245354af28dcaef) to any openocd repo and then rebuild the project.

choncba, 2017-03-20:
I already add these two patches to the openocd repo  //git.code.sf.net/p/openocd/code openocd-code (latest version), (I directly replace the at91samd.c file with these changes included), but after compile, and run openocd, the restore command give me an error, something missing?

facchinm, 2017-03-20:
Could you paste here the error?

choncba, 2017-03-20:
`openocd.cfg error: 'at91samd restore' unrecognized command`

facchinm, 2017-03-20:
I didn't read the sentence `I directly replaced`; it won't work, since other patches made their way into release between 0.9 and 0.10. You'll need to backport the patches manually (by only changing the relevant parts)

choncba, 2017-03-20:
OK, I will try and let you know

choncba, 2017-03-24:
Well, after getting the openocd source code in its most recent version, I edited the **src/flash/nor/at91samd.c** file directly to add the restore command, then compile and install openocd normally.
Then I can program this with this script:

```
Source [find interface / raspberrypi2-native.cfg]
Transport select swd
 
Set CHIPNAME at91samd21g18
Source [find target / at91samdXX.cfg]
 
Adapter_nsrst_delay 100
Adapter_nsrst_assert_width 100
 
Init
Targets
Reset halt

At91samd bootloader 0
Program samd21_sam_ba_arduino_mkrzero.bin verify
At91samd bootloader 8192
Reset
Shutdown
```

The result:

```
Open On-Chip Debugger 0.10.0 + dev-00093-g6b2acc0-dirty (2017-03-23-19: 15)
Licensed under GNU GPL v2
For bug reports, read
        Http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
Srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Adapter speed: 400 kHz
Cortex_m reset_config sysresetreq
Adapter_nsrst_delay: 100
Adapter_nsrst_assert_width: 100
Info: BCM2835 GPIO JTAG / SWD bitbang driver
Info: SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info: clock speed 400 kHz
Info: SWD DPIDR 0x0bc11477
Info: at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName Type Endian TapName State
------------------ ---------- ------ -------------- ----------------
 0 * at91samd21g18.cpu cortex_m little at91samd21g18.cpu halted
Target halted due to debug-request, current mode: Handler HardFault
XPSR: 0x41000003 pc: 0xfffffffe msp: 0xffffffd8
Target halted due to debug-request, current mode: Handler HardFault
XPSR: 0x41000003 pc: 0xfffffffe msp: 0xffffffd8
** Programming Started **
Auto erase enabled
Info: SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
Wrote 16384 bytes from file samd21_sam_ba_arduino_mkrzero.bin in 0.653277s (24,492 KiB / s)
** Programming Finished **
** Verify Started **
Verified 6404 bytes in 0.526342s (11,882 KiB / s)
** Verified OK **
Target halted due to breakpoint, current mode: Handler HardFault
XPSR: 0x61000003 pc: 0x2000002e msp: 0xffffffd8
Shutdown command invoked

```
Then I performed the restore with this script:

```
Source [find interface / raspberrypi2-native.cfg]
Transport select swd
 
Set CHIPNAME at91samd21g18
Source [find target / at91samdXX.cfg]
 
Adapter_nsrst_delay 100
Adapter_nsrst_assert_width 100
 
Init
Targets
Reset halt
At91samd restore
Reset
```

The result:

```
Open On-Chip Debugger 0.10.0 + dev-00093-g6b2acc0-dirty (2017-03-23-19: 15)
Licensed under GNU GPL v2
For bug reports, read
        Http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
Srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Adapter speed: 400 kHz
Cortex_m reset_config sysresetreq
Adapter_nsrst_delay: 100
Adapter_nsrst_assert_width: 100
Info: BCM2835 GPIO JTAG / SWD bitbang driver
Info: SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info: clock speed 400 kHz
Info: SWD DPIDR 0x0bc11477
Info: at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName Type Endian TapName State
------------------ ---------- ------ -------------- ----------------
 0 * at91samd21g18.cpu cortex_m little at91samd21g18.cpu running
Target halted due to debug-request, current mode: SysTick Handler
XPSR: 0x6100000f pc: 0x00000752 msp: 0x20007fb0
```

Even so, windows still does not detect the usb device.

facchinm, 2017-03-27:
The last logs look good, however if it doesn't solve the USB enumeration I doubt I can help much more. If the W25 and companion hardware are exactly the same as a MKR1000 and the software is the same (and the fuses are back to a safe value), there must be something wrong going on at a deeper level. I'm sorry but I believe you should try to contact someone at Atmel/Microchip explaining what is going on, probably they have a clue (or they can provide you another debug method).

closed 2017-03-27 by facchinm