arduino/ArduinoCore-samd #229: no windows driver for MKRFOX1200

elektroHorst, 2017-04-27:
It cant get the new MKRFOX1200 board running on Windows 7, because of missing drivers.

pacav69, 2017-04-27:
Drivers

On Windows, drivers are needed to allow the board communication. These drivers will be installed automatically when adding the core.
ref: https://www.arduino.cc/en/Guide/MKRFox1200#toc10

Installing additional Arduino Cores
https://www.arduino.cc/en/Guide/Cores

elektroHorst, 2017-04-27:
I installed the SAMD core.
But when I have a look into the installed arduino-samd.inf, there is no MKRFOX1200 board.
There are only drivers for: MKR1000, MKRZero, SmartEverything Fox
But the hardware ID of the  MKRFOX1200 is different.

pacav69, 2017-04-27:
i did a search for MKRFOX1200 and it returned this setting

[image]
is this what you found?

akash73, 2017-04-27:
Hi all,
thanks for head up on this.
unfortunately our tests skipped the windows 7 systems.
we will add the windows 7 drivers support later today, sorry for any inconvenience.
I will keep you updated here once will be available

thanks!

pacav69, 2017-04-28:
looks like the driver has been updated or being worked on.

akash73, 2017-04-28:
Hi all!
thanks for the patience. :)
SAMD core 1.6.15 with support for Windows 7 is available through the library manager.

closed 2017-04-28 by akash73