arduino/ArduinoCore-samd #232: Remove external 32KHz crystal from Arduino Zero (SAMD21)

FredMeester, 2017-05-04:
In my current project, I want to remove the external 32KHz oscillator crystal from the Arduino Zero board. 

I re-configured the startup sequence to use the 32KHz internal oscillator of the SAMD21. In the comments of the two *.c files mentioned below, the startup sequence is explained. Step 1 from the SAM-BA bootloader and startup.c from the Arduino core is removed. In step 2, XOSC32K is changed with OSCULP32K. This is the internal low power 32K oscillator.

https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/board_init.c 
https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/startup.c 

Unfortunately, there is still power measured from the external crystal. Are there more steps needed to remove the external crystal?

facchinm, 2017-05-04:
Hi @FredMeester ,
you only need to define CRYSTALLESS (as you can see here https://github.com/adafruit/ArduinoCore-samd/blob/master/boards.txt#L145 ) to avoid using the 32K crystal in the sketch code. However, the bootloader still expects it and since we have no boards without the quartz we also don't provide a suitable bootloader. From the [Adafruit](https://github.com/adafruit/ArduinoCore-samd) repo you can grab the Circuit Playground M0 bootloader which doesn't need the external oscillator.
Please reopen the issue if you need further assistance.

closed 2017-05-04 by facchinm

FredMeester, 2017-05-05:
@facchinm Thanks for pointing me to boards.txt where CRYSTALLESS is declared. I configured the bootloader for using the internal oscillator and did a make build. It seems the crystal is still working. 
[image]
[image]

mirage335, 2018-05-30:
As mentioned in a previous bug report, there is a working bootloader here.
https://github.com/arduino/ArduinoCore-samd/issues/157
https://github.com/kliment/ArduinoCore-samd

Been a while since I last used it though, just now taking another look.

tinkerBOY-git, 2019-12-11:
I just tried the bootloader at https://github.com/kliment/ArduinoCore-samd/blob/master/bootloaders/zero/samd21_sam_ba.bin to flash a custom board i made with atsamd21g18a without crystal and this [Wemos Mini board](https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=wemos+samdi21+mini&_sacat=0). I used Atmel Studio 7 and the bootloader uploads just fine.  I can also upload Arduino sketches just fine but uploaded sketches won't work unless i put in a external 32khz crystal.

Do I have to re-compile the bootloader?