arduino/ArduinoCore-samd #234: SoftwareSerial examples for Arduino M0 will not compile

control48, 2017-05-05:
IDE: 1.8.2
Installation: C:\Program Files (x86)\Arduino\portable\packages\arduino\hardware\samd\1.6.15\libraries\SoftwareSerial\SoftwareSerial.cpp and \SoftwareSerial.h  (from github from samd hardware)

Application : SoftwareSerialExample.ino (the original example)

Errors:

sketch\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX(uint8_t)':

SoftwareSerial.cpp:174: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile PORT_OUT_Type*' in assignment

   _transmitPortRegister = portOutputRegister(port);

                         ^

sketch\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setRX(uint8_t)':

SoftwareSerial.cpp:186: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile PORT_IN_Type*' in assignment

   _receivePortRegister = portInputRegister(port);

                        ^

exit status 1
cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile PORT_OUT_Type*' in assignment

What is wrong in the data type/ struc/?? that breaks this function?  Attempted to port srv code base to samd base and the broken samd version does not work for serial I/O.  Does anyone have a working example or modified .cpp / .h set that works?

facchinm, 2017-05-08:
Hi @control48 ,
can you post the link you used yo download the SoftwareSerial library? The SAMD core doesn't include it by default so you likely downloaded it from a different location. 
Keep in mind that SAMD architecture virtually doesn't need a SoftwareSerial library since any SERCOM port can be configured as Serial on a plethora of pins (see [this](https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports?view=all) post for a complete explanation on how to configure an additional serial port and the available pins)

control48, 2017-05-08:
I believe the solution has been posted by pharaohamps in

https://forum.arduino.cc/index.php?topic=341054.30
where the two lines in SoftwareSerial.cpp:

169:  _transmitPortRegister = portOutputRegister(port);
182:  _receivePortRegister = portInputRegister(port);

should be corrected to:

169:  _transmitPortRegister = (volatile PORT_OUT_Type*)portOutputRegister(port);
182:  _receivePortRegister = (volatile PORT_IN_Type*)portInputRegister(port);

Now if someone would correct this source and add the correct keywords.txt and library.properties so that the library (under \Arduino\portable\packages\arduino\hardware\samd\1.6.15\libraries\SoftwareSerial\ is the likely place for the install) could be used as other libraries.  Thanks to that person.
BTW, I am testing the above fix not for at least a program with the library change will at least compile w/o errors.

control48, 2017-05-08:
I used the following link to download the original files.

https://github.com/arduino-org/Arduino/tree/1364019868e2ad044c87dd0f143052a92eebda98/hardware/arduino/samd/libraries/SoftwareSerial

BlueNalgene, 2018-03-04:
@control48 's changes seem to have worked for me as well.  Since this is not a default file, it should be noted that instances of `<SoftwareSerial.h>` in the .cpp file and in your .ino must be changed to `"SoftwareSerial.h"` to include the local files.

garageeks, 2019-07-29:
Hi to all,
I'd like to ask if anyone had stability issues with this SAMD21 SoftwareSerial library.
I'm using it in a project where all SERCOM pins have been assigned to other peripherals.
Therefore I connected it to pin PA06, PA07 (Pins 8,9 on a Sparkfun SAMD21 Mini)
When I use a simple example sketch, it works with no issues, but when incorporated into a code that uses a SIM800 modem and Adafruit Fona library, the software hangs randomly, but always at the same point (When receiving HTTP data from the modem through the serial buffer)
The result is a complete lockup that happens after 5-60 minutes of uptime.

Thanks

Skulduggery0, 2024-03-19:
Does anyone still have the library, arduino didn't include it when the migrated from arduino-org to arduino